openwrt/luci #3870: Name validation broken in dchp.js

resadent, 2020-04-10:
Hi. I realised that I cannot insert manualy DNS forwardings in luci  dhcp and dns section, it says `Expecting: valid hostname` every time I try to insert a dns server. I realised the problem should be in this function from `modules/luci-mod-network/htdocs/luci-static/resources/view/network/dhcp.js`:

`function validateServerSpec(sid, s)`

However my regexp knowledge is not enough to fix it.

jow-, 2020-04-10:
Works for me. What input is failing exactly?

resadent, 2020-04-11:
> Works for me. What input is failing exactly?
I was trying with 9.9.9.9, 9.9.9.11, 192.168.10.40. None of these seemed to work. I guess Id have to try a new build? Mine is from april 6.

closed 2020-04-11 by jow-

jow-, 2020-04-11:
Thanks. I was only testing the `/domain/ipaddr` notation, not the IPs directly.

resadent, 2020-04-11:
> Thanks. I was only testing the `/domain/ipaddr` notation, not the IPs directly.

No problem, happy to help :).

commit 137db1c4 (2020-04-11): luci-mod-network: dhcp.js: fix validation logic

The `server` option allows plain IPs besides the `/domain/addr` format.

Fixes: #3870