openwrt/luci #3874: luci stats for adsl/vdsl

ezplanet, 2020-04-10:
Hi, I am looking to collect stats for ADSL and VDSL values like ES, SES, LATN, SATN, SNR and Data Rate. Is there a plug-in already for luci stats?
If not, is it possible to add a feature request?

Kaye101, 2020-04-10:
how do i run for Luci latest update.....

ysc3839, 2020-04-10:
What are these data from?

jow-, 2020-04-10:
Existing no, possible likely. I do not own any *DSL hardware, nor do I have access to any, so you need to tell us exactly what to plot, and where/how to obtain the raw data (e.g. the CLI commands).

ezplanet, 2020-04-10:
The dsl data elements are provided by:
/etc/init.d/dsl_control lucistat

I am looking to feed some of these into luci stats

ezplanet, 2020-04-10:
If you do not have a xDSL modem, here is an example of what 
/etc/init.d/dsl_control lucistat returns:

local dsl={}
dsl.atuc_vendor_id="Infineon 178.6"
dsl.atuc_system_vendor_id="45,43,49,20,74,65,6C,65"
dsl.chipset="Lantiq-VRX200"
dsl.firmware_version="5.9.1.4.0.7"
dsl.api_version="4.17.18.6"
dsl.xtse1=0x0
dsl.xtse2=0x0
dsl.xtse3=0x0
dsl.xtse4=0x0
dsl.xtse5=0x0
dsl.xtse6=0x0
dsl.xtse7=0x0
dsl.xtse8=0x2
dsl.xtse_s="0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x2"
dsl.annex_s="B"
dsl.line_mode_s="G.993.2 (VDSL2)"
dsl.profile=6
dsl.profile_s="17a"
dsl.line_state_num=0x801
dsl.line_state_detail="showtime_tc_sync"
dsl.line_state="UP"
dsl.errors_fec_near=105121650
dsl.errors_fec_far=4
dsl.errors_es_near=9
dsl.errors_es_far=343
dsl.errors_ses_near=1
dsl.errors_ses_far=0
dsl.errors_loss_near=0
dsl.errors_loss_far=354
dsl.errors_uas_near=109
dsl.errors_uas_far=109
dsl.errors_hec_near=0
dsl.errors_hec_far=0
dsl.errors_crc_p_near=3
dsl.errors_crc_p_far=0
dsl.errors_crcp_p_near=0
dsl.errors_crcp_p_far=0
dsl.power_mode_num=0
dsl.power_mode_s="L0 - Synchronized"
dsl.latency_down="8000"
dsl.latency_up="0"
dsl.latency_num_down="8.0 ms"
dsl.latency_num_up="0.0 ms"
dsl.latency_s_down="Interleave"
dsl.latency_s_up="Fast"
dsl.data_rate_down=32388000
dsl.data_rate_up=7157000
dsl.data_rate_down_s="32.388 Mb"
dsl.data_rate_up_s="7.157 Mb"
dsl.line_attenuation_down="24.5"
dsl.line_attenuation_up="34.3"
dsl.noise_margin_down="5.3"
dsl.noise_margin_up="6.2"
dsl.signal_attenuation_down="22.0"
dsl.signal_attenuation_up="33.7"
dsl.actatp_down="4.6"
dsl.actatp_up="4.7"
dsl.max_data_rate_down=31358976
dsl.max_data_rate_up=7402140
dsl.max_data_rate_down_s="31.358 Mb"
dsl.max_data_rate_up_s="7.402 Mb"
dsl.line_uptime=4883
dsl.line_uptime_s="1h 21m 23s"
return dsl

avoid data elements ending with "_s" as there is always a corresponding numeric data element without "_s" suffix.

solidus1983, 2020-04-20:
DSL/VDSL user here, each modem has different ways of getting this data. I have a Raspi 3B+ setup that does this with a software called DSLStats which works on some modems quite well. However I am now running a mongoose web server on my current modem to do the same thing.

[image]
[image]
[image]
[image]
[image]

However in both instances you need to be able to telnet into the modem to get the info which is normally locked out by the OEM's, so the only way I can think of is through SNMP but that has informational limits.

jow-, 2020-04-20:
@solidus1983 - I believe the issue here is graphing the output of various `/etc/init.d/dsl_control lucistat`metrics in collectd, not checking stats of external DSL modems.

solidus1983, 2020-04-20:
@jow So i take it they are after placing everything to a graph?

If thats the case then using prometheus and grafana with the prometheus-node-exporter-lua-ltq-dsl package should do the trick.

jow-, 2020-04-20:
Yeah, or writing ~ 30 lines of JS to display it in the existing collectd framework on the router itself.

solidus1983, 2020-04-20:
There is that @jow- 

I just thought someone was asking for the rabbit hole as there is many binarys for DSL stats and switches.

```
adsl
adslcfg
adslcmd
adslctl
adslphxcmd
xdslcfg
xdslcmd
xdslctl
xdsl
xdslctl0
xdslctl1
```

Then there is the whole telnet or ssh issues on most OEM modems where ssh is normal disabled and telnet is either locked down completely or disabled to where only TR069 can enable it with a config update and reboot.

ezplanet, 2020-04-30:
> @jow So i take it they are after placing everything to a graph?
> 
> If thats the case then using prometheus and grafana with the prometheus-node-exporter-lua-ltq-dsl package should do the trick.

That is a bit of an overkill...

stav242, 2021-01-10:
fy, i found this other link.. which is doing something simialar on zyxel devices..
https://gist.github.com/squarewf/79d632ed5482b160978ef4ce3494b95b

seems the dsl_control `lucistat` output option might make this simpler...  if i get time i might try to get it working.

jow-, 2022-12-19:
The `luci-mod-dsl` module got merged with https://github.com/openwrt/luci/commit/cd48d766add74df60b6c6a882defa859b311ec97 and #5844 is pending for collectd integration, please follow up there.

closed 2022-12-19 by jow-