openwrt/luci #3895: Latest luci rerpc error!!

zxlhhyccc, 2020-04-11:
[image]

fuqiang03, 2020-04-11:
[image]

a-kibou, 2020-04-11:
[image]
[image]

jow-, 2020-04-11:
Sorry, I can't read this and you didn't mention the URL it occurs on. I cannot reproduce the DHCP and DNS problem in the initial post. Ubus code 6 means that permission denied. Make sure you run the latest LuCI including https://github.com/openwrt/luci/commit/43e872871a85dce70509c3c97c10e9c61788c378

closed 2020-04-11 by jow-

jow-, 2020-04-11:
The ubus code 4 issue on the main status display is addressed with 65d673f4671215f2907baebc8d6860f64ba13307, I forgot to test on a unit without wireless. You can work around the problem with `touch /etc/config/wireless`.