openwrt/luci #3903: New build - LUCI crashing, but router is running otherwise...

dk4dk4, 2020-04-12:
I cannot launch LUCI on my router after a fresh build: 
OpenWRT SNAPSHOT, r12911-e66becb490

Hitting router IP (192.168.1.1) on browser gives me:
Sat Apr 11 23:12:26 2020 daemon.err uhttpd[10282]: /usr/lib/lua/luci/ccache.lua:70: error loading module 'luci.controller.advanced_reboot' from file '/usr/lib/lua/luci/controller/advanced_reboot.lua':
Sat Apr 11 23:12:26 2020 daemon.err uhttpd[10282]:      /usr/lib/lua/luci/controller/advanced_reboot.lua:46: ')' expected (to close '(' at line 45) near 'alt_partition_unmount'
Sat Apr 11 23:12:26 2020 daemon.err uhttpd[10282]: stack traceback:
Sat Apr 11 23:12:26 2020 daemon.err uhttpd[10282]:      [C]: in function 'loader'
Sat Apr 11 23:12:26 2020 daemon.err uhttpd[10282]:      /usr/lib/lua/luci/ccache.lua:70: in function </usr/lib/lua/luci/ccache.lua:61>
Sat Apr 11 23:12:26 2020 daemon.err uhttpd[10282]:      [C]: in function 'require'
Sat Apr 11 23:12:26 2020 daemon.err uhttpd[10282]:      /usr/lib/lua/luci/dispatcher.lua:985: in function 'createindex'
Sat Apr 11 23:12:26 2020 daemon.err uhttpd[10282]:      /usr/lib/lua/luci/dispatcher.lua:1096: in function 'createtree'
Sat Apr 11 23:12:26 2020 daemon.err uhttpd[10282]:      /usr/lib/lua/luci/dispatcher.lua:639: in function 'menu_json'
Sat Apr 11 23:12:26 2020 daemon.err uhttpd[10282]:      /usr/lib/lua/luci/dispatcher.lua:734: in function 'dispatch'
Sat Apr 11 23:12:26 2020 daemon.err uhttpd[10282]:      /usr/lib/lua/luci/dispatcher.lua:441: in function </usr/lib/lua/luci/dispatcher.lua:440>

F12 - on this page:
Failed to load resource: the server responded with a status of 500 (Internal Server Error)

The build before this today the LuCI interface is stuck and a message "Loading view..." appears at any page I navigate and then hangs indefinitely.  This made me do another build today.  I cannot get into LUCI at all now...

hnyman, 2020-04-12:
Cc @stangri

closed 2020-04-12 by hnyman

hnyman, 2020-04-12:
@dk4dk4 
you can manually edit your live router if you want to fix this quickly:
Simply add a ")" to end of lines 45 and 54 in /usr/lib/lua/luci/controller/advanced_reboot.lua

See the fix as a diff in 2c812fbd5

dk4dk4, 2020-04-12:
Thanks this resolved the issue.

commit 2c812fbd (2020-04-12): luci-app-advanced-reboot: fix syntax

Add the missing ")" to lines 45 and 54.
Fixes 75a4a80c1dd
Fixes #3903
commit 30af84e3 (2020-04-12): luci-app-advanced-reboot: fix syntax

Add the missing ")" to lines 45 and 54.
Fixes 75a4a80c1dd
Fixes #3903

(cherry picked from commit 2c812fbd58fd7399969185cc501bdba561d98d95)
commit 103da80a (2020-04-12): luci-app-advanced-reboot: fix syntax

Add the missing ")" to lines 45 and 54.
Fixes 75a4a80c1dd
Fixes #3903

(cherry picked from commit 2c812fbd58fd7399969185cc501bdba561d98d95)