openwrt/luci #3911: make  error!

snakwu, 2020-04-14:
bash: % 86400: syntax error: operand expected (error token is "% 86400")               
/home/snakwu/openwrt-master/openwrt/feeds/luci/luci.mk:277: *** Package/luci-i18n-acces
scontrol-zh-cn is missing the VERSION field.  Stop.

hnyman, 2020-04-14:
Anything special with you build environment?
Normal OpenWrt repo with LuCI as a feed?
Did you compile full firmware, or just a module?

(The LuCI version string logic was modfied a few days ago, but it should work ok in normal conditions.)

@jow- 
see also
https://forum.openwrt.org/t/luci-mk-package-luci-base-is-missing-the-version-field-stop/60547

Just wondering, which conditions trigger the error.

jow-, 2020-04-14:
I guess it happens for LuCI packages not being in Git.

snakwu, 2020-04-14:
> Anything special with you build environment?
> Normal OpenWrt repo with LuCI as a feed?
> Did you compile full firmware, or just a module?
>
> (The LuCI version string logic was modfied a few days ago, but [...]

I added some plug-ins to compile, it is normal before the fields have been modified, now it is wrong!

snakwu, 2020-04-14:
> I guess it happens for LuCI packages not being in Git.

I think this is also possible

closed 2020-04-14 by jow-

snakwu, 2020-04-14:
> I guess it happens for LuCI packages not being in Git.

The problem has not been solved!
make[3]: Entering directory '/home/snakwu/openwrt-master/openwrt/package/lean/luci-app-accesscontrol'
bash: % 86400: syntax error: operand expected (error token is "% 86400")
/home/snakwu/openwrt-master/openwrt/feeds/luci/luci.mk:288: *** Package/luci-i18n-accesscontrol-zh-cn is missing the VERSION field. Stop.
make[3]: Leaving directory '/home/snakwu/openwrt-master/openwrt/package/lean/luci-app-accesscontrol'
time: package/lean/luci-app-accesscontrol/compile#0.83#0.14#1.74
package/Makefile:111: recipe for target 'package/lean/luci-app-accesscontrol/compile' failed
make[2]: *** [package/lean/luci-app-accesscontrol/compile] Error 2
make[2]: Leaving directory '/home/snakwu/openwrt-master/openwrt'
package/Makefile:107: recipe for target '/home/snakwu/openwrt-master/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/snakwu/openwrt-master/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/snakwu/openwrt-master/openwrt'
/home/snakwu/openwrt-master/openwrt/include/toplevel.mk:218: recipe for target 'world' failed
make: *** [world] Error 2

jow-, 2020-04-14:
Provide the output of the following commands:
```
cd /home/snakwu/openwrt-master/openwrt/package/lean/luci-app-accesscontrol
git log -1
git log -1 --format="%ct %h" --abbrev=7 -- ':(exclude)po'
```

snakwu, 2020-04-14:
> Provide the output of the following commands:
> 
> ```
> cd /home/snakwu/openwrt-master/openwrt/package/lean/luci-app-accesscontrol
> git log -1
> git log -1 --format="%ct %h" --abbrev=7 -- ':(exclude)po'
> ```

commit 7c2e0fa5865406d0f2d1987bdd2180c1d1913bf2 (HEAD -> master, origin/master, origin/HEAD)
Author: Daniel Golle <daniel@makrotopia.org>
Date:   Tue Apr 14 16:11:05 2020 +0100

    procd: jail fixes and improvements
    
     32c717e jail: only mess with rootfs if CLONE_NEWNS was set
     b275a62 instance: harmonize instance API
     511fd97 jail: make /proc more secure
     4953b7c jail: mount /sys read-only
     a4d6442 jail: replace /etc/resolv.conf with symlink in extroot+overlay
     a4cc165 jail: always mount /dev as additional tmpfs
    
    Signed-off-by: Daniel Golle <daniel@makrotopia.org>
snakwu@hassio-server:~/openwrt-master/openwrt/package/lean/luci-app-accesscontrol$ git log -1 --format="%ct %h" --abbrev=7 -- ':(exclude)po'
1586877366 7c2e0fa
snakwu@hassio-server:~/openwrt-master/openwrt/package/lean/luci-app-accesscontrol$

snakwu, 2020-04-15:
> Provide the output of the following commands:
> 
> ```
> cd /home/snakwu/openwrt-master/openwrt/package/lean/luci-app-accesscontrol
> git log -1
> git log -1 --format="%ct %h" --abbrev=7 -- ':(exclude)po'
> ```

Compilation passed, but the settings interface of some plugins was lost!

snakwu, 2020-04-15:
@a-kibou  请不要把其他分支的问题相互引用，以免混乱！

a-kibou, 2020-04-15:
> @a-kibou 请不要把其他分支的问题相互引用，以免混乱！

抱歉，那我應該如何刪除？

commit 9d8e99f9 (2020-04-14): build: gracefully handle non-Git source trees

Fixes: #3911
Fixes: 59fe4086f ("build: luci.mk: drop support for non-Git SCMs")
commit e6f77d5d (2020-04-14): build: luci.mk: gracefully handle missing or unversioned po subdirectories

Fixes: #3911
Fixes: 9d8e99f9b build: gracefully handle non-Git source trees
commit 07056577 (2020-04-14): build: luci.mk: gracefully handle missing or unversioned po subdirectories

Fixes: #3911
Fixes: 9d8e99f9b build: gracefully handle non-Git source trees