openwrt/luci #3912: Physical Interface Removal Error

TJT-JTJ, 2020-04-14:
When removing a physical interface from a network interface in luci error is produced:

 > An error occurred while saving the form:
RPC call to uci/delete failed with ubus code 4: Resource not found
  at ClassConstructor.handleCallReply []((http://192.168.1.1/luci-static/resources/rpc.js?v=git-19.292.31773-cc35206:16:3)[](url))

jow-, 2020-04-14:
Can you describe the steps leading to this in more detail? I have a hunch that this already addressed by https://github.com/openwrt/luci/commit/55cb5a753b94e2eee049546a2af374bb6129d4c4 by I want to be sure.

It might also help to attach your current `/etc/config/network` file.

TJT-JTJ, 2020-04-14:
The steps in the first post are each click I make to produce the error.

Here's my config for network
`config interface 'loopback'
        option ifname 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'

config interface 'vlan2'
        option type 'bridge'
        option proto 'static'
        option ipaddr '192.168.1.1'
        option netmask '255.255.255.0'
        option ip6assign '60'
        option ifname 'lan1 lan2'

config interface 'wan'
        option proto 'pppoe'
        option password 'PPPoEuser'
        option ipv6 'auto'
        option username 'PPPoEpass'
        option ifname 'wan'

config switch
        option name 'switch0'
        option reset '1'
        option enable_vlan '1'

config switch_vlan    
        option device 'switch0'
        option vlan '2'   
        option ports '6t 4'  
        option vid '2'           
                                  
config rule6               
        option mark '0x65'
        option lookup '101'
                      
config interface 'vlan2'   
        option proto 'static'
        list ipaddr '10.0.0.1/24'  
        option delegate '0'           
        option type 'bridge' 
        option ifname 'lan3 lan4'
                           
config switch_vlan          
        option device 'switch0'  
        option vlan '3'           
        option vid '3'    
        option ports '5t 1 0'       
                           
config interface 'MODEM'
        option proto 'static'
        option ipaddr '192.168.100.2'
        option netmask '255.255.255.0'
        option ifname 'wan'  `

jow-, 2020-04-14:
Thank you, that helped to track down the issue.

commit 6ae1cd8f (2020-04-14): luci-base: uci.js: do not issue malformed uci/delete requests

Fixes: #3912

(cherry picked from commit ec3a202b69b5bf5785c0785b2ac39efbe5cae2b7)
closed 2020-04-14 by jow-