openwrt/luci #3913: Feature Request: sshtunnel LUCI interface

lars18th, 2020-04-14:
Hi,

The package `sshtunnel` is very useful to create a poor man's VPN tunnels with the omnipresent SSH protocol. The advantage over other alternative like wireguard is that this doesn't require server support. Futhermore this package works like a charm inside OpenWRT. So it will be desirable to have a LUCI module to configure it.

I hope someone will want to implement it.
Thank you!

systemcrash, 2023-12-12:
Closed via #6424

closed 2023-12-12 by systemcrash

lars18th, 2023-12-18:
> Hi @lars18th @marazmarci here was added the [luci-app-sshtunnel](https://github.com/openwrt/luci/tree/master/applications/luci-app-sshtunnel) Please try and test.

Thank you @stokito . It works!

Another question: Please could you make a new PR with this? https://github.com/openwrt/packages/pull/20163