openwrt/luci #3914: the new luci no switch config display

ptpt52, 2020-04-14:
on page `/cgi-bin/luci/admin/network/switch`

device: mt7621
ubnt-erx-sfp

I build mt7621 with swconfig and ralink-eth support and everything work but switch luci page show no config

`/etc/config/network` has config for switch and network work fine
```
config switch
	option name 'switch0'
	option reset '1'
	option enable_vlan '1'

config switch_vlan
	option device 'switch0'
	option vlan '1'
	option ports '0 1 2 3 6t'

config switch_vlan
	option device 'switch0'
	option vlan '2'
	option ports '4 6t'
```

/etc/board.json
```
{
	"model": {
		"id": "ubiquiti,edgerouterx-sfp",
		"name": "UBNT-ERX-SFP"
	},
	"switch": {
		"switch0": {
			"enable": true,
			"reset": true,
			"ports": [
				{
					"num": 1,
					"role": "lan"
				},
				{
					"num": 2,
					"role": "lan"
				},
				{
					"num": 3,
					"role": "lan"
				},
				{
					"num": 4,
					"role": "lan"
				},
				{
					"num": 0,
					"role": "wan"
				},
				{
					"num": 6,
					"device": "eth0",
					"need_tag": false,
					"want_untag": false
				}
			],
			"roles": [
				{
					"role": "lan",
					"ports": "1 2 3 4 6t",
					"device": "eth0.1"
				},
				{
					"role": "wan",
					"ports": "0 6t",
					"device": "eth0.2"
				}
			]
		}
	},
	"network": {
		"lan": {
			"ifname": "eth0.1",
			"protocol": "static",
			"macaddr": "74:8xxxxxxxxxxx"
		},
		"wan": {
			"ifname": "eth0.2",
			"protocol": "dhcp",
			"macaddr": "74:8xxxxxxxxxxx"
		}
	},
	"gpioswitch": {
		"poe_power_port0": {
			"name": "PoE Power Port0",
			"pin": 496,
			"default": 0
		},
		"poe_power_port1": {
			"name": "PoE Power Port1",
			"pin": 497,
			"default": 0
		},
		"poe_power_port2": {
			"name": "PoE Power Port2",
			"pin": 498,
			"default": 0
		},
		"poe_power_port3": {
			"name": "PoE Power Port3",
			"pin": 499,
			"default": 0
		},
		"poe_power_port4": {
			"name": "PoE Power Port4",
			"pin": 500,
			"default": 0
		}
	}
}
```

ptpt52, 2020-04-14:
it seems
`var switchSections = uci.sections('network', 'switch');` get null
in file : luci-static/resources/view/network/switch.js

jow-, 2020-04-14:
Is `swconfig` installed?

ptpt52, 2020-04-14:
swconfig installed
and
`swconfig list` get the resault:
```
Found: switch0 - mt7530
```

jow-, 2020-04-14:
What is reported by `uci show network` and `ubus call session list` ?

ptpt52, 2020-04-14:
very long output

```
ubus call session list
{
        "ubus_rpc_session": "00000000000000000000000000000000",
        "timeout": 0,
        "expires": 0,
        "acls": {
                "access-group": {
                        "unauthenticated": [
                                "read"
                        ]
                },
                "ubus": {
                        "luci": [
                                "getFeatures"
                        ],
                        "session": [
                                "access",
                                "login"
                        ]
                }
        },
        "data": {

        }
}
{
        "ubus_rpc_session": "77805123f7575a34e4c9e869a6d1c2d7",
        "timeout": 3600,
        "expires": 3586,
        "acls": {
                "access-group": {
                        "luci-app-ddns": [
                                "read",
                                "write"
                        ],
                        "luci-app-firewall": [
                                "read",
                                "write"
                        ],
                        "luci-app-macvlan": [
                                "read",
                                "write"
                        ],
                        "luci-app-mwan3": [
                                "read",
                                "write"
                        ],
                        "luci-app-natcap": [
                                "read",
                                "write"
                        ],
                        "luci-app-nft-qos": [
                                "read",
                                "write"
                        ],
                        "luci-app-openvpn": [
                                "read",
                                "write"
                        ],
                        "luci-app-opkg": [
                                "read",
                                "write"
                        ],
                        "luci-app-sqm": [
                                "read",
                                "write"
                        ],
[... 498 lines cut ...]
                        "upnpd": [
                                "read",
                                "write"
                        ],
                        "wireless": [
                                "read",
                                "write"
                        ],
                        "wizard": [
                                "read",
                                "write"
                        ],
                        "xwan": [
                                "read",
                                "write"
                        ]
                }
        },
        "data": {
                "token": "f9aef8d876a55ea71180be24138c9909",
                "username": "root"
        }
}

```

ptpt52, 2020-04-14:
and
```
uci show network
network.loopback=interface
network.loopback.ifname='lo'
network.loopback.proto='static'
network.loopback.ipaddr='127.0.0.1'
network.loopback.netmask='255.0.0.0'
network.globals=globals
network.globals.ula_prefix='fd84:cb37:4bc1::/48'
network.globals.packet_steering='1'
network.lan=interface
network.lan.type='bridge'
network.lan.ifname='eth0.1 eth0.200'
network.lan.proto='static'
network.lan.ipaddr='192.168.15.1'
network.lan.netmask='255.255.255.0'
network.lan.ip6assign='60'
network.wan6=interface
network.wan6.proto='dhcpv6'
network.wan6.ifname='@wan'
network.wan6.reqaddress='try'
network.wan6.reqprefix='auto'
network.@switch[0]=switch
network.@switch[0].name='switch0'
network.@switch[0].reset='1'
network.@switch[0].enable_vlan='1'
network.@switch_vlan[0]=switch_vlan
network.@switch_vlan[0].device='switch0'
network.@switch_vlan[0].vlan='1'
network.@switch_vlan[0].ports='0 1 2 6t'
network.@switch_vlan[1]=switch_vlan
network.@switch_vlan[1].device='switch0'
network.@switch_vlan[1].vlan='2'
network.@switch_vlan[1].ports='4 6t'
network.@switch_vlan[2]=switch_vlan
network.@switch_vlan[2].device='switch0'
network.@switch_vlan[2].vlan='200'
network.@switch_vlan[2].ports='3t 6t'
network.wan=interface
network.wan.proto='pppoe'
network.wan.ifname='eth0.2'
network.wan.metric='40'
network.wan.username='user'
network.wan.password='password'
network.wan.keepalive='5 5'
network.wan.mtu='1492'
network.wan.ipv6='1'
network.wan1=interface
network.wan1.proto='static'
network.wan1.ifname='eth0.2'
network.wan1.ipaddr='192.168.17.254'
network.wan1.netmask='255.255.255.0'
network.wan1.gateway='192.168.17.1'
network.wan1.metric='41'

```

jow-, 2020-04-14:
Can't see any obvious problem and unable to reproduce using an ar71xx build on the only swconfig capable device I have lying around here.

ptpt52, 2020-04-14:
yep very strange it only happens on mt7621，i will take more inspection.
thank you for your help.

ptpt52, 2020-04-14:
@jow- 
hi, I found the point:

my device is ubnt-erx-sfp, it has no wireless
but after I touch a file `touch /etc/config/wireless`
the switch page work again
so it is related missing `/etc/config/wireless`, a bug in some js

you may reproduce this issue by `rm /etc/config/wireless`

ptpt52, 2020-04-14:
related on this commit: bd713f870a974d46b214d07d85ac4a895d85a122

ptpt52, 2020-04-14:
and this commit: 65d28a4ac92a4806b5ad836a2dbbe8bc4e87207b

ptpt52, 2020-04-14:
and resolve by modify this:
```
diff --git a/modules/luci-base/htdocs/luci-static/resources/uci.js b/modules/luci-base/htdocs/luci-static/resources/uci.js
index e6582b3e2..5dd98f5b2 100644
--- a/modules/luci-base/htdocs/luci-static/resources/uci.js
+++ b/modules/luci-base/htdocs/luci-static/resources/uci.js
@@ -31,8 +31,7 @@ return baseclass.extend(/** @lends LuCI.uci.prototype */ {
                object: 'uci',
                method: 'get',
                params: [ 'config' ],
-               expect: { values: { } },
-               reject: true
+               expect: { values: { } }
        }),
 
        callOrder: rpc.declare({
```

commit 0e22d4d8 (2020-04-15): luci-base: network.js: don't fail loading network config on missing wireless

Fixes: #3914
closed 2020-04-15 by jow-