openwrt/luci #3916: adblock，Hope to be able to support custom filtering rules！thx

snakwu, 2020-04-15:
adblock，Hope to be able to support custom filtering rules！thx

dibdot, 2020-04-15:
This is an issue tracker, not a wish list ... ;-)
I do not plan to expose such things to LuCI. Currently you have more than 40 pre-configured lists plus local black- and whitelist. If you insist to change things, feel free to change the JSON objects in `/etc/adblock/adblock.sources.gz`, see the readme for details.

closed 2020-04-15 by dibdot

snakwu, 2020-04-16:
> This is an issue tracker, not a wish list ... ;-)
> I do not plan to expose such things to LuCI. Currently you have more than 40 pre-configured lists plus local black- and whitelist. If you insist to [...]

@dibdot No matter how many filtering rules built in can not meet everyone's needs, custom filtering rules can meet everyone's needs, thank you!