openwrt/luci #3917: Hi, @ jow- About luci.mk, please take a look!

snakwu, 2020-04-15:
After applying this fix https://github.com/openwrt/luci/commit/0705657747e31a0884ee7268515f04ee2684ae8e, the compilation passed, but the added plug-in, the settings interface is missing, open blank, thank you!

jow-, 2020-04-15:
No idea what this is supposed to mean. What plug-in, what settings interface?

snakwu, 2020-04-15:
> No idea what this is supposed to mean. What plug-in, what settings interface?

 added luci-app-accesscontrol, this plugin, and then the compilation error https://github.com/openwrt/luci/issues/3911, under your fix https://github.com/openwrt/luci/commit / 0705657747e31a0884ee7268515f04ee2684ae8e, it can be successfully compiled, but in the UI of luci-app-accesscontrol, open the blank!

snakwu, 2020-04-15:
@jow- 

[image]
[image]

[image]
[image]

qiuzi, 2020-04-15:
@snakwu 
https://github.com/openwrt/luci/commit/43e872871a85dce70509c3c97c10e9c61788c378

snakwu, 2020-04-15:
> @snakwu 
> https://github.com/openwrt/luci/commit/43e872871a85dce70509c3c97c10e9c61788c378

？？？

qiuzi, 2020-04-15:
@snakwu nft-qos已经修复了 缺少ACL规则文件的问题造成配置文件无法读取

snakwu, 2020-04-15:
> @snakwu nft-qos已经修复了 缺少ACL规则文件的问题造成配置文件无法读取

还有很多旧的插件出现这种况，如何解决？

snakwu, 2020-04-15:
> @snakwu nft-qos已经修复了 缺少ACL规则文件的问题造成配置文件无法读取

特别是用了其它人修改的插件，基本都会出现这个问题

jow-, 2020-04-15:
It would be great if you could use english so I can follow. As for the question on how it can be solved... a simple fix would be shipping a wildcard ACL that simply allows access to all UCI files, like it has been before. It would look like this:

```
{
	"allow-full-uci-access": {
		"description": "Allow access to all UCI files",
		"read": {
			"uci": [ "'*" ]
		},
		"write": {
			"uci": [ "'*" ]
		}
	}
}
```

Not sure how to best handle it though, it should be made optional somehow since it basically undermines the entire purpose of the fine grained ACLs which is to allow for some proper permission management eventually.

snakwu, 2020-04-15:
> It would be great if you could use english so I can follow. As for the question on how it can be solved... a simple fix would be shipping a wildcard ACL that simply allows access to all UCI files, like it has been before. It would look like this:
> 
> ```
> {
> 	"allow-full-uci-access": {
> 		"description": "Allow access to all UCI files",
> 		"read": {
> 			"uci": [ "*" ]
> 		},
> 		"write": {
> 			"uci": [ "*" ]
> 		}
> 	}
> }
> ```
> 
> Not sure how to best handle it though, it should be made optional somehow since it basically undermines the entire purpose of the fine grained ACLs which is to allow for some proper permission management eventually.
> 
> Maybe I should add it to `luci-compat` since this is required for most "legacy" (Lua, non-JS) apps.

Now even old plug-ins have this problem.

qiuzi, 2020-04-15:
@jow-  Now luci is becoming more and more complex, which is not conducive to developers getting started quickly. And affect the framework

jow-, 2020-04-15:
@qiuzi - I am sorry but these changes are needed to ensure proper security. I'll move forward with adding the wildcard ACL to luci-compat later.

Closing this ticket now since packages in this tree are fixed. Packages maintained elsewhere need to start adding proper ACL rules.

closed 2020-04-15 by jow-

qiuzi, 2020-04-15:
@jow- Thank you for everything

snakwu, 2020-04-15:
> @qiuzi - I am sorry but these changes are needed to ensure proper security. I'll move forward with adding the wildcard ACL to luci-compat later.
> 
> Closing this ticket now since packages in this tree are fixed. Packages maintained elsewhere need to start adding proper ACL rules.

@jow- thank you very much