openwrt/luci #3924: Protocol column is blank when using an OpenVPN config file using proto tcp

jamesmacwhite, 2020-04-17:
I have a couple of OpenVPN configurations written as: 

```
config openvpn 'openvpn_example'
        option config '/etc/openvpn/example.ovpn'
        option enabled '1'
```

For some reason the ovpn configuration files using proto tcp, does not display 'tcp' in the protocol column within LuCI, where as the UDP OpenVPN configs do. This is an example screenshot:

[image]

dibdot, 2020-04-17:
could you provide the content of your example.ovpn? Thanks!

jamesmacwhite, 2020-04-17:
One of the TCP configs looks like this with sensitive stuff redacted:

```
client
dev tun2
proto tcp
verify-x509-name "REDACTED"
route remote_host 255.255.255.255 net_gateway
resolv-retry infinite
nobind
persist-key
persist-tun
<ca>
------ CA HERE ------
</ca>
<cert>
------ CERT HERE ------
</cert>
<key>
------ KEY HERE ------
</key>
auth-user-pass /etc/openvpn/authfile
cipher AES-128-CBC
auth SHA256
comp-lzo no
route-delay 4
verb 5
reneg-sec 0
remote xxx.xxx.xxx.xxx 8443
```

dibdot, 2020-04-18:
fixed with d4fef2cc647b2381c195f2e417a883b1c0adfb15

closed 2020-04-18 by dibdot

jamesmacwhite, 2020-04-18:
@dibdot Thanks for the fix!