openwrt/luci #3926: luci-app-openvpn: Add a list of currently connected clients

mastergregor, 2020-04-18:
It would be nice to have a list of currently connected clients for each of the OpenVPN servers. Currently one can pull this from a status file of the given server, but it is a bit cumbersome. Maybe this listing can be added to an existing openvpn page, or a separate status page under Status drop down menu.

If a separate status page is created, maybe status of other openvpn instances could be displayed as well, like client connection info?

Just an idea from someone coming to OpenWRT from DD-WRT.

jow-, 2020-05-05:
Can you provide an example of the status file?

mastergregor, 2020-05-06:
Sure thing jow, I am available for testing if you wish as well. I tried to dig a bit into it, but page layout and composition is a bit complex, and I would need to get familiar with the luci architecture first.

In any case, attached is the shell output from the router illustrating the status file. First "cat" is with no clients connected, second one is after a PSK client connects. It appears to me that it is a comma delimited info file, with line breaks. Note that default location of the status file can be overridden when starting the OpenVPN server. By default, OpenVPN updates the status file every 1 minute. This is also configurable though.

======================================
root@router:/# cat /tmp/run/openvpn.CriticalMainVPN.status
TITLE,OpenVPN 2.4.7 arm-openwrt-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD]
TIME,Tue May  5 22:40:53 2020,1588736453
HEADER,CLIENT_LIST,Common Name,Real Address,Virtual Address,Virtual IPv6 Address,Bytes Received,Bytes Sent,Connected Since,Connected Since (time_t),Username,Client ID,Peer ID
HEADER,ROUTING_TABLE,Virtual Address,Common Name,Real Address,Last Ref,Last Ref (time_t)
GLOBAL_STATS,Max bcast/mcast queue length,0
END
root@router:/#
root@router:/#
root@router:/#
root@router:/# cat /tmp/run/openvpn.CriticalMainVPN.status
TITLE,OpenVPN 2.4.7 arm-openwrt-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD]
TIME,Tue May  5 22:41:33 2020,1588736493
HEADER,CLIENT_LIST,Common Name,Real Address,Virtual Address,Virtual IPv6 Address,Bytes Received,Bytes Sent,Connected Since,Connected Since (time_t),Username,Client ID,Peer ID
CLIENT_LIST,GregorLaptop,192.168.1.5:53408,192.168.2.6,,8016,5075,Tue May  5 22:41:21 2020,1588736481,UNDEF,,0
HEADER,ROUTING_TABLE,Virtual Address,Common Name,Real Address,Last Ref,Last Ref (time_t)
ROUTING_TABLE,192.168.2.6,GregorLaptop,192.168.1.5:53408,Tue May  5 22:41:22 2020,1588736482
GLOBAL_STATS,Max bcast/mcast queue length,0
END
root@router:/#
======================================= 

I am also attaching a text file with the same info. Let me know how I can help.

[OpenVPNStatus.txt](https://github.com/openwrt/luci/files/4584738/OpenVPNStatus.txt)

Solution40, 2022-06-09:
Hi All,

Would be nice to have this option. Let me know how i can help!

sebabordon, 2023-06-23:
I've made a script and status page for this based on contributions from different sources on the openwrt forum and uploaded it to github.
https://github.com/sebabordon/openwrt-tools/

sebabordon, 2023-06-23:
I've made a script for this based on contributions from different sources on the openwrt forum and uploaded it to github 
https://github.com/sebabordon/openwrt-tools/

schunckt, 2025-07-21:
If somebody is interested in, here [https://github.com/sebabordon/openwrt-tools/issues/2](url) I placed a simple extension of the Status Overview page which shows all details pulled from the OpenVPN status file.
Code is based on @sebabordon openwrt-tools