openwrt/luci #3931: luci-app-adblock result[0].trim is not a function

shuaibird, 2020-04-19:
[image]

dibdot, 2020-04-19:
Just update your luci and rpcd packages.

closed 2020-04-19 by dibdot

singalen, 2020-05-23:
Just happened to me.

```
~# opkg update
...
# opkg upgrade luci rpcd
Upgrading luci on root from git-20.136.49537-fb2f363-1 to git-20.142.51595-0c6fd62-1...
Downloading http://downloads.openwrt.org/releases/19.07.3/packages/arm_cortex-a15_neon-vfpv4/luci/luci_git-20.142.51595-0c6fd62-1_all.ipk
Configuring luci.
```
The package is freshly installed, not even present in the menu. Maybe something is missing in the data?

kernc, 2020-05-29:
After a fresh install, a _reboot_ seems to be required.

spicem, 2020-06-22:
same issue. reboot not helping