openwrt/luci #3940: Feature request: luci-app-https-dns-proxy could add dropdown options for "1.1.1.1 for Families"

rajasdevel, 2020-04-20:
Cloudflare has recently added service that blocks malware and adult content: [1.1.1.1 for Families](https://developers.cloudflare.com/1.1.1.1/1.1.1.1-for-families/).

It would be great if the option to select those services appeared on the "Resolver" drop-down menu, in addition to standard Cloudflare resolver.

jow-, 2020-05-05:
@stangri

stangri, 2020-05-05:
I've added these to the luci app, however it doesn't look like the https-dns-proxy service works with those two new cloudflare resolvers. It won't be a quick fix.

n-thumann, 2020-10-04:
Works fine me: I can select all three CF DNS Services (General, Familiy & Security) and the all work as expected ✌️

stangri, 2020-10-05:
Sorry, this should have been closed when principal package moved to RFC8484 support.

closed 2020-10-05 by stangri