openwrt/luci #3942: luci-app-nut, Issues with Material theme

arrmo, 2020-04-21:
Hi,

Trying to find where the files are on the router itself, to try to fix this, but in the meantime ... hopefully the pics below demonstrate the issue. It's specific to the material theme, and for nut-monitor.

1) Bootstrap
[image]

2) Material
[image]

Basically, the select box needs to be taller - it sort of works right now, but really messes with selection. The list "leaks" outside the box (when trying to select).

Thanks!

hnyman, 2020-04-25:
Material theme is pretty much unmaintained.

arrmo, 2020-04-25:
Ahh, OK - gotcha. I thought it was sort of the "newer" one, but seems that's not the case. I'll try to dig, thanks.

hnyman, 2020-04-25:
> I thought it was sort of the "newer" one

Well, that depends on your perspective.

When I started with OpenWrt ten years ago, there was just "openwrt theme", and somewhat later in 2012 came the current Bootstrap theme.

Material was introduced in 2015, but original author stopped contributing in 2017, so the theme is decaying.

Newly introduced well-maintained theme is luci-theme-openwrt-2020.

arrmo, 2020-04-25:
Thanks for the info - much appreciated!

Will give that one a try.

closed 2020-04-25 by arrmo

arrmo, 2020-04-25:
OK, added this to my build, really like it - thanks again!