openwrt/luci #3948: luci-app-openvpn: ask for password interactively

hadmut, 2020-04-21:
Hi, 

many openvpn servers nowadays use some sort of one-time-password, e.g. with some kind of token with a display or mobile phone app. 

luci-app-openvpn (openwrt as an openvpn client) currently does not really cope with changing password. One would have to 'edit' the connection every time and change the password, then quickly start the connection. 

would be nice of a connection could be configured to query a onetime password when pressing 'start' (or having a text field near the button). 

regards

kduvekot, 2020-05-06:
I think it would be good to seperate a few things here:

one-time-password that you have to enter during connection as "the" password

or configuration when there is a username/password AND "an other factor" that you can have to enter also (the "static-challenge" option in the openvpn configuration.

ideally you should be able to add any of those via a GUI option, or provide defaults that can be saved.

kduvekot, 2020-05-06:
also .. as far as I can see .. the default setting for OpenVPN on OpenWRT is to not include the "management' option .. and that means that static-challenge is not working with the default openvpn-openssl client.

https://github.com/openwrt/openwrt/blob/master/package/network/services/openvpn/Config-openssl.in#L23

so that results in the following error message:

`Unrecognized option or missing or extra parameter(s) in /etc/openvpn/connection.ovpn:131: static-challenge (2.4.7)`