openwrt/luci #3957: luci-app-sqm: can't change settings

resadent, 2020-04-23:
Hi. I have openwrt from 19th april, I have seen no changes related to this since then so I figured Id post this.

I can't change settings in sqm via luci interface, ie. I can't click in save/apply button, save or reset button. This didn't happen at first but now I can't fix it no matter what I try, so I'm changing the settings via cli.
[image]

hnyman, 2020-04-23:
It should have been fixed in master on April 20th.

https://git.openwrt.org/?p=project/luci.git;a=commitdiff;h=74da73b012a7ac271d683f11d98de170215ae661;hp=760763a7b5da42860c6c2d63478859bb02fa9104#patch25

Remove and reinstall sqm.

Note that there has been buildbot trouble, so the updated package might not yet be there.

closed 2020-04-23 by resadent

resadent, 2020-04-23:
I confirm its fine with latest build compiled a few moments ago.