openwrt/luci #3963: Theme Lost on Firmware Upgrade (Flash)

arrmo, 2020-04-24:
Hi,

Most settings are preserved when a new image is flashed, but it seems the theme (Design) setting is lost. Is this expected?

Thanks!

hnyman, 2020-04-25:
> the theme (Design) setting is lost. Is this expected?

Theme is not just a setting, but an installable package
Losing is
* expected, if you install a release or snapshot build and you have not used Bootstrap, the default theme.
* unexpected if you have built a personal firmwre image where the theme is built-in.

closed 2020-04-25 by hnyman

arrmo, 2020-04-25:
Hi,

It's a personal build, where the theme is built-in ... so the unexpected case 😆. I agree with you, if an installed packaged, this would / should go away (but in this case, seems it shoudl stay).

Thanks!