openwrt/luci #3969: OpenWRT 19.07.1 -> 19.07.2 upgrade UI  issues

sl0n, 2020-04-25:
Hi,

I just ran 19.07.1 -> 19.07.2 upgrade on TP-Link Archer C2600, and noticed that clicking on "Status -> System Log" triggers the following error:

`Failed to execute call dispatcher target for entry '/admin/status/syslog'.
The called action terminated with an exception:
/usr/lib/lua/luci/template.lua:74: Failed to load template 'admin_status/syslog'.
Error while parsing template '/usr/lib/lua/luci/view/admin_status/syslog.htm':
No such file or directory
stack traceback:
	[C]: in function 'error'
	/usr/lib/lua/luci/template.lua:74: in function '__init__'
	/usr/lib/lua/luci/util.lua:65: in function 'Template'
	/usr/lib/lua/luci/template.lua:27: in function 'render'
	/usr/lib/lua/luci/controller/admin/status.lua:33: in function </usr/lib/lua/luci/controller/admin/status.lua:31>
	(tail call): ?`

Actions "System-> Routes" and "System-> Kernel log" are triggering a similar error. 

List of luci related packages installed:

`# opkg list_installed  | grep luci
liblucihttp-lua - 2019-07-05-a34a17d5-1
liblucihttp0 - 2019-07-05-a34a17d5-1
luci - git-20.113.71444-50c05f3-1
luci-app-firewall - git-20.113.71444-50c05f3-1
luci-app-opkg - git-20.113.71444-50c05f3-1
luci-app-vpn-policy-routing - git-20.115.28077-2aeb83a-71
luci-app-wireguard - git-20.113.71444-50c05f3-1
luci-base - git-20.113.71444-50c05f3-1
luci-compat - git-20.113.71444-50c05f3-1
luci-lib-ip - git-20.113.71444-50c05f3-1
luci-lib-jsonc - git-20.113.71444-50c05f3-1
luci-lib-nixio - git-20.113.71444-50c05f3-1
luci-mod-admin-full - git-20.113.71444-50c05f3-1
luci-mod-network - git-20.113.71444-50c05f3-1
luci-mod-status - git-20.113.71444-50c05f3-1
luci-mod-system - git-20.113.71444-50c05f3-1
luci-proto-ipv6 - git-20.113.71444-50c05f3-1
luci-proto-ppp - git-20.113.71444-50c05f3-1
luci-proto-wireguard - git-20.113.71444-50c05f3-1
luci-ssl-openssl - git-20.113.71444-50c05f3-1
luci-theme-bootstrap - git-20.113.71444-50c05f3-1
rpcd-mod-luci - 20191114
`

zosorock, 2020-04-28:
I am getting the same error on a Linksys EA4500. I am very new to OpenWRT though and just upgraded all packages that were available with a loop and `opkg upgrade pkgname`.

jow-, 2020-04-28:
You need to reboot or run `rm -r /tmp/luci-*`.

closed 2020-04-28 by jow-