openwrt/luci #3984: No translation for Software section

ptipti23, 2020-05-01:
No full translation for "Software" section. Everything always in English
[image]

stangri, 2020-05-01:
Just to clarify, when you say "everything is in english", do you mean package names and package descriptions?

ptipti23, 2020-05-01:
Menu item, tabe titles. Not package names

stangri, 2020-05-01:
It's a community-translated project, you can contribute to translations too: https://hosted.weblate.org/projects/openwrt/luci/ru/

ptipti23, 2020-05-01:
Thanks for link, but is says that translation completed100%. "Software" section was a part of luci-base module in 18X OpenWRT, but now I cannot find this phrases in pot file.

stangri, 2020-05-01:
You are right, weblate does show that luci-base is 100% translated to Russian, but according to the String Status on that same page not all of the translations have been approved.

Also, from reading [this](https://github.com/openwrt/luci/issues/3183), I'm not sure if weblate translations are making it into 18.06 tree.

@aparcar could you please advise? 

@aparcar can you please shed some light on the issue?

ptipti23, 2020-05-01:
In 18.06 translation is OK. I am talking about LuCi 19X an other daily snapshots. In [this POT file](https://github.com/openwrt/luci/blob/master/modules/luci-base/po/templates/base.pot) missins msg about Software not only for Russian, but for all languages

stangri, 2020-05-01:
Ah, right, I've completely misread your earlier message. I'm guessing it's due to some code being rewritten in javascript. Also, I do see some traces of the resources for that page in [this file](https://github.com/openwrt/luci/blob/master/modules/luci-base/po/en/base.po). Please wait for more seasoned team members to chime in.

aparcar, 2020-05-01:
All strings seem to be [translated](https://hosted.weblate.org/zen/openwrt/luciapplicationsopkg/ru/), looking at the strings from [luci-app-opkg](https://github.com/openwrt/luci/tree/master/applications/luci-app-opkg). Is it still unavailable in the latest snapshot?

ptipti23, 2020-05-02:
It seems opkg was translated only 10 hours ago (luci/applications/luci-app-opkg/po/ru/opkg.po)

hnyman, 2020-05-02:
> It seems opkg was translated only 10 hours ago

I ran the treewide language template sync yesterday after noticing your issue, and that has probably generated the necessary templates.

Ps.
@stangri 
But that i18n-sync got me into major trouble, as weblate didn't like some strings with quoted quotemarks in luci-app-vpn-policy-routing package 

`applications/luci-app-vpn-policy-routing/po/pl/vpn-policy-routing.po | Syntax error on line 287`

Apparently weblate parsing did not like quoted quotemark at the end of a line
```
+"uruchomić następujące pliki użytkownika. Zobacz <a href=\"%s\" target=\""
+"_blank\">README</a> aby uzyskać szczegółowe informacje."
```
It might be better to leave the URL href with quotes outside the translatable string.

stangri, 2020-05-02:
@hnyman just to confirm, is it choking on `\"%s\"`? I have that in 4 of my luci apps, should I send/merge a single PR for a "tree-wide" change?

hnyman, 2020-05-02:
I think that weblate actually chke by the escaped " as the last character of the row. Placed there automatically by the gettext line length routines. Pure coincidence, but took a while to clarify, as there was a merge error plus a new "parsing error".

It might be better to have the whole HREF item in %s, so that the translatable string would not have any escaped \" characters. That would simplify the string quite aignificantly.

stangri, 2020-05-02:
> It might be better to have the whole HREF item in %s, so that the translatable string would not have any escaped " characters. That would simplify the string quite aignificantly.

Thanks, that's what I'm currently testing locally -- moving the opening and closing tags into %s.

hnyman, 2020-05-02:
That might be the most clear solution

(there are only a few apps that have translated href URLs in LuCI, so I guess we have so far escaped weblate's quirks, but yesterday there happened to be a character combination that it did not like.

jow-, 2020-05-02:
Did you install `luci-i18n-opkg-ru`?

stangri, 2020-05-03:
@jow- Would it make sense/be possible to make the `luci-i18n-opkg-xx` a dependency of `luci-i18n-base-xx`?

ptipti23, 2020-05-03:
if it possible, place include "luci-i18n-firewall-xx"

ptipti23, 2020-05-03:
> Did you install `luci-i18n-opkg-ru`? The software management is provided by `luci-app-opkg` which is an extra package, it is not covered by the base translations.

Yes, but still missing translation. I'm looking to git source at "https://downloads.openwrt.org/snapshots/packages/mipsel_24kc/luci/", it seems packages updated, I reinstall `luci-i18n-opkg-ru`, but nothing happends. Opkg still in english

jow-, 2020-05-03:
> Would it make sense/be possible to make the luci-i18n-opkg-xx a dependency of luci-i18n-base-xx?

No, since it would break the modularity.

ptipti23, 2020-05-03:
Firewall and now opkg is default apps. So if you want your language by default, you have to install 3 language packages? May be you create virtial package, ex. `luci-i18-default-xx` and this would install -base-opkg-firewall?

jow-, 2020-05-03:
Yes.

castillofrancodamian, 2020-05-03:
> Firewall and now opkg is default apps. So if you want your language by default, you have to install 3 language packages? May be you create virtial package, ex. `luci-i18-default-xx` and this would install -base-opkg-firewall?

Those apps are default, not their translations.

ptipti23, 2020-05-03:
Well, so be it.
I already collected po2lmo and autonomously created opkg.ru.lmo, put it in the /usr/lib/lua/luci/i18n folder, reloaded the router ... But again, I don’t see the Russian language. Why?

ptipti23, 2020-05-04:
Today openwrt snapshots were updated. Every thing is OK, `opkg` now in Russian. Thanks alot, Issue solved

closed 2020-05-04 by hnyman