openwrt/luci #3987: rrdtool.js TypeError: opts.data is undefined

rmandrad, 2020-05-02:
with luci-app-statistics_git-20.113.70804-9d02ef1_all ...  can't display processor statistics it is stuck on "loading data" ... debugging on the browser shows that opts.data is undefined on rrdtool.js

hnyman, 2020-05-02:
Hmm. I think that the CPU related pages work ok for me.  

CPU load (processor) and system load load ok with (git-20.119.49022-81effc1) / OpenWrt SNAPSHOT r13136-ad27c133eb.

Which page exactly?  
Which router & architecture?
Normal master build?

Test with other browser, clear browser cache  etc.

rmandrad, 2020-05-02:
WRT3200ACM - own build from master; on two browsers (firefox & chrome) same issue 

I am able to display data for other plugins like wifi,entropy ... sensors.

Navigating to Statistics -> Graphs -> Processor

debuggin on Edge I get a slightly different msg but one is pointing to undefined/null reference

HTTP400: BAD REQUEST - The request could not be processed by the server due to invalid syntax.
(XHR)GET - https://192.169.99.1:18443/ubus/?1588400740595

SCRIPT5007: SCRIPT5007: Unable to get property 'types' of undefined or null reference
graphs (54,1)

20: Unable to get property 'types' of undefined or null reference
rrdtool.js (53,1)

hnyman, 2020-05-02:
For me the processor (CPU) tab works quite normally on WRT3200ACM with 
LuCI Master (git-20.119.49022-81effc1)
OpenWrt SNAPSHOT r13136-ad27c133eb

hnyman, 2020-05-02:
I will close this, as it does not seem to be a global bug.

If you figure out any details, feel free to add here.

Ps.
> (XHR)GET - https://192.169.99.1:18443/ubus/?1588400740595

I assume that you know that 192.169.x.x is not a private IP addressing area, but common routable internet.
https://en.wikipedia.org/wiki/Private_network

closed 2020-05-02 by hnyman