openwrt/luci #3989: Unable to set Wireless Country code in Luci

vietanhbui2000, 2020-05-02:
I'm unable to set Wireless Country code in Luci with recent master build. The only way i can set it is by directly editing /etc/config/wireless. But after editing, if any changes is made in Luci, the Wireless Country code option will get deleted with uci command immediately.
[image]

castiel652, 2020-05-02:
This also happened to me.

zxlhhyccc, 2020-05-02:
> 我无法使用最新的主版本在Luci中设置无线国家/地区代码。我可以设置它的唯一方法是直接编辑/ etc / config / wireless。但是，在编辑后，如果在Luci中进行了任何更改，则将使用uci命令立即删除“无线国家/地区代码”选项。
> 下面的屏幕截图是我在Luci中进行一些更改后发生的情况。
> [image]

No problem with mine, please pull...

vietanhbui2000, 2020-05-02:
> > 我无法使用最新的主版本在Luci中设置无线国家/地区代码。我可以设置它的唯一方法是直接编辑/ etc / config / wireless。但是，在编辑后，如果在Luci中进行了任何更改，则将使用uci命令立即删除“无线国家/地区代码”选项。
> > 下面的屏幕截图是我在Luci中进行一些更改后发生的情况。
> > [image]
> 
> No problem with mine, please pull...

I'm sorry, pull what?

commit e3ef463c (2020-05-02): luci-base: ui.js: fix input value reading for select widgets

Fixes: #3989
Fixes: 81effc111 ("luci-base: ui.js: assume select widget by default for ListValue widget")
closed 2020-05-02 by jow-