openwrt/luci #4020: baseclass is not defined In statistics page

Ansuel, 2020-05-06:
I have this error when i try to open the graph section in the statistics page

hnyman, 2020-05-06:
I guess that you have ran into the same issue that I stumbled upon two weeks ago:
You have disabled javascript minification, right?

I discussed that privately two weeks ago with @jow- 

--------------------------
(Message from 20 April 2020, so the date references reflect that)

Good news and bad news ;-)

I got completely baffled, as after "make dirclean" also my R7800 build started to throw fit on that Statistics/graphs page.

But I think that I have now figured it out:  the reason is JavaScript minify, which is enabled by default.

Yesterday evening I turned JS minify off, as I wanted to be able to edit the SQM cake stats JS easier.
So, my builds today have been done without JS minify. 
`# CONFIG_LUCI_JSMIN is not set`

That was the actual change today that triggered to bug for me.

Firmware built yesterday with LuCI 8cc82a8 with standard JS minify does not cause that "baseclass is not defined" error,
Today's master fdce990  without JS minify causes the error also in R7800.
Today's master 398b198 with standard JS minify works ok, again.

Apparently the current LuCI code only works with JS minify enabled, and causes that error if JS minify gets turned off and LuCI is built from a commit after
https://git.openwrt.org/?p=project/luci.git;a=commitdiff;h=94a0c571810da25043ca3599561bd9b369c0c65e;hp=125916f2f4603acf980acd074aca55d0b528f34b

which removed the old "?v=git-19.292.31773-cc35194" tag you added in December (by purpose?).  That is just a guess based on the WRT3200ACM regression testing (with JS minify off all the time).

Does it make any sense to you?

I rebuilt up-to-date WRT3200ACM with JS minify, and now also it works ok.

The bad news is that apparently at least that one Statistics/Graphs page really requires JS minify.  I have no idea why that one page experiences that error with unminified source, but possibly it is because of the dynamic content of the page (evaluation of the available data & plugins etc.) & and the overview/details functionality for some of the plugins.

Ansuel, 2020-05-06:
yes I also have js min disabled
the strange part is that other js script that also use baseclass doesn't suffer of this problem... 
Could be caused by a wrong way of load the dynamic content in the parent js script ?

hnyman, 2020-05-06:
I have no javascript skills, so I have not tried to debug it.

I guessed earlier to jow that it might be the overview/detail tab functionality implemented with 72da3c4c5509a4fe9d779e03d76a97854022fafe

Statistics is the only app that uses it, I think. So that might be  a partial reason, but I have no idea why JS minification would prevent hitting a bug there.

Ansuel, 2020-05-06:
This is funny.... the problem is in the license comments of the uptime module...
it does work in min js becuase minified files have stripped comments AHAHAHHA

Ansuel, 2020-05-06:
@jow- i don't know if this is a bug....

putting `"` in the comments cause this error

Ansuel, 2020-05-06:
By using `\"`  this is resolved but still... shouldn't commends be completely ignored?

hnyman, 2020-05-06:
I will remove the " from there in any case.
But strange that it would cause the error.

hnyman, 2020-05-06:
Possibly solved with 4f80f9542f

Ansuel, 2020-05-06:
I don't know... I would see that as a workaround... Think it's better to wait Jow

hnyman, 2020-05-06:
Yeah, that is just a workaround and does not prevent future cases. 
The question why a quotemark there would cause it, is the interesting one.

Ansuel, 2020-05-06:
Fixed this but i don't think this will get accepted as a bit overkill for a very specific case... Will propose anyway...
(the problem is in the Luci require function that parse all the code and search for `"` or `'` without skipping any comment)

commit 6be8f8db (2020-05-29): luci-base: luci.js: properly skip comments when finding require tokens

Fixes: #4020, #4022, #4111
commit dd9624d3 (2020-05-29): luci-base: luci.js: properly skip comments when finding require tokens

Fixes: #4020, #4022, #4111

(cherry picked from commit 6be8f8dbaee90a8a7e8bc87350f21793b9aed35c)
closed 2020-05-29 by jow-