openwrt/luci #4021: Question about LuCI rpc and ubus

hagaygo, 2020-05-06:
Hi

I am not sure this is the proper place to ask this question , if not, please refer me to the correct one.

I am building an android app that able to show OpenWRT status/data from OpenWRT devices on the network.

I managed to use current relevant LuCI commands and call them in my app.

What i am currently missing is a command that will allow me to have a translation of wifi client mac address to ip(4).

There are cases it does return it , but most of my clients i get only MAC address (i have 1 router setup without wifi and 2 ap device , all openwrt)

I am looking for output similar to "arp -a" command.

I understand that LuCI uses ubus cli command , i tried to find a cli command with ubus that will return this data, but without any luck.

Thanks in advance for any reply.

Is this supported by ub

jow-, 2020-05-06:
Since the AP devices will not handle IP traffic but only forward layer 2 data frames, they usually have no ARP information about connected clients until you implement some kind of flood pinging or reverse ARP resolution.

This is outside of the scope of LuCI though, it would be something that needs to be implemented (and enabled) in OpenWrt first. Reading the data is straight forward, you could e.g. use the ubus `file` namespace with the `read` procedure to retrieve `/proc/net/arp` or you could use the `luci-rpc` / `getHostHints` method to dump information about known hosts.

closed 2020-05-06 by jow-

hagaygo, 2020-05-06:
Thanks for your help.