openwrt/luci #4040: Luci’s menu disappeared

9pr19, 2020-05-10:
Hello,

I’ve lost Luci’s menu in the last few weeks:
[image]

With web browser network monitoring I’ve noticed the following error:

> 404 Not Found
>
>Sorry, the object you requested was not found.
>Unable to dispatch: /cgi-bin/luci/admin/menu

[image]

 System:
> OpenWrt 19.07.2, r10947-65030d81f3

Luci packages:
> liblucihttp-lua - 2019-07-05-a34a17d5-1
> liblucihttp0 - 2019-07-05-a34a17d5-1
> luci-app-ddns - 2.4.9-7
> luci-app-firewall - git-20.130.24486-07e6cfa-1
> luci-app-opkg - git-20.130.24486-07e6cfa-1
> luci-app-statistics - git-20.130.24486-07e6cfa-1
> luci-app-uhttpd - git-20.130.24486-07e6cfa-1
> luci-base - git-20.130.24486-07e6cfa-1
> luci-compat - git-20.130.24486-07e6cfa-1
> luci-lib-ip - git-20.130.24486-07e6cfa-1
> luci-lib-ipkg - git-20.130.24486-07e6cfa-1
> luci-lib-iptparser - git-20.130.24486-07e6cfa-1
> luci-lib-jsonc - git-20.130.24486-07e6cfa-1
> luci-lib-nixio - git-20.130.24486-07e6cfa-1
> luci-mod-admin-full - git-20.130.24486-07e6cfa-1
> luci-mod-network - git-20.130.24486-07e6cfa-1
> luci-mod-status - git-20.130.24486-07e6cfa-1
> luci-mod-system - git-20.130.24486-07e6cfa-1
> luci-proto-ipv6 - git-20.130.24486-07e6cfa-1
> luci-proto-ppp - git-20.130.24486-07e6cfa-1
> luci-theme-bootstrap - git-20.130.24486-07e6cfa-1
> rpcd-mod-luci - 20191114

tverlaan, 2020-05-11:
I'm experiencing the same issue. I see this in my browser console, it might help:
```
Uncaught (in promise) SyntaxError: Unexpected token < in JSON at position 0
    at JSON.parse (<anonymous>)
    at ClassConstructor.json (luci.js?v=git-20.130.24486-07e6cfa:32)
    at ClassConstructor.eval (ui.js?v=git-20.130.24486-07e6cfa:215)
```

jow-, 2020-05-11:
`rm -r /tmp/luci-*` or reboot your router.

closed 2020-05-11 by jow-

9pr19, 2020-05-11:
I removed the folders then login into Luci and everything is back.
@jow- thank you.

rammstein9910, 2023-11-22:
I have tried what @jow- advised, but as a plan B... I used, and it worked `opkg install --force-reinstall luci`

olenikolai, 2024-09-04:
None of the above worked for me, but this guide at [Techoverflow](https://techoverflow.net/2021/05/04/how-i-fixed-openwrt-luci-error-etc-config-luci-seems-to-be-corrupt-unable-to-find-section-main/) fixedt it for me. The problem was a bad rpcd configuration, fixed it with the commands:
cp /rom/etc/config/rpcd /etc/config/rpcd
service uhttpd restart