openwrt/luci #4062: QoS over Nftables Enable/Disable Checkbox Not Working

firewire10000, 2020-05-17:
**Device:** Linksys WRT1900ACSv2
**Software Versions:** OpenWrt SNAPSHOT r13209-e31e5c4ea4 / LuCI Master git-20.131.46247-2cea1a4
**Kernel version:** 4.19.115

**Step to reproduce**
1. Check the box `Limit Enable` checkbox and click the `Save & Apply` button at the bottom.
2. Under `Static QoS-Download Rate` and `Static Upload Rate` click the `Add` button.
3. Enter the IP address of the device you want to limit bandwidth on in the IP address field. Use a computer as you will need this to test the throttled bandwidth with an internet speed test.
4. Enter a rate lower than your WAN internet speeds in the Rate field for Download and Upload sections and click the `Save & Apply` button at the bottom.
5. On the throttled computer do a speed test at your favourite and most reliable WAN speed test site/tool. I use Ookla SPEEDTEST. You will notice that your speeds have indeed been limited.
6. Go back to QoS over Nftables and now uncheck the `Limit Enable` checkbox and click the `Save & Apply` button at the bottom.
7. Now do a second speed test on the same site/tool and you should find the limited still applies, despite the `Limit Enable` checkbox being unchecked.

jow-, 2020-05-17:
@rosysong

koshev-msk, 2020-12-19:
**S/V:** OpenWrt 19.07.5 r11257-5090152ae3 / LuCI openwrt-19.07 branch git-20.351.44718-4edcf38
**Devices:** ZBT-WE826-T2 (16M), ZBT-WE3526 (16M)
**Kernel:** 4.14.209
Similar a problem. I checking boxes, but bandwith rules does not apply.