openwrt/luci #4065: luci-app-statistics: weekly graphs are sometimes plotted with very few data points

Schneehuhn, 2020-05-17:
Sometimes, weekly graphs are plotted with very few data points used on the x-axis. Daily and hourly graphs look fine.

[image]

There is nothing suspicious about the rrd database itself.

hnyman, 2020-05-17:
I have noticed the same since the move to JS based graphs, but the graphs have always corrected themselves after vising a few other apps & tabs. I think that I have also seen that happening also with the daily graph.

Apparently some times the timescale index gets somehow skewed by one, and the data for the next longer time period is used instead of the correct one (thus reducing the amount of data points).

So far I have not been able to figure out the reason.

Schneehuhn, 2020-05-17:
I've noticed as well that the issue corrects itself after a few hours. When the bug manifests itself, all weekly graphs are affected. What does not work is emptying the image cache in `/tmp/rrdimg`, the recalculated graphs look just as before.

jow-, 2020-05-17:
Try playing with the `-s` and `-e` value here when this happens: https://github.com/openwrt/luci/blob/master/applications/luci-app-statistics/htdocs/luci-static/resources/statistics/rrdtool.js#L337

The `-e` was an addition I made when converting to the client side, the intent here was avoiding an open gap at the graph end when displaying the hourly graph. Maybe it has a negative effect on long time range renderings.

hnyman, 2020-05-17:
Thanks for the hint about the change you made in the JS transition.  I will try to figure it out the next time.

Comparing master to the 19.07 code, there is that -e addition, but apparently there is not much more relevant difference.

The -e is always "NOW - 60 seconds". The time scale makes no difference, so even in the weekly data, the required scale is upto this minute (instead of scaled like NOW - 2 hours ago).

Based on that, a guess of the possible situation:

* Possibly rrdtool automatically determines the suitable data table based on the length of the required period. 
* Earlier it was given NOW-week as start and automatic end(?)
* Now it gets NOW-week and NOW-60sec.  Maybe that time period is slightly longer than the data found in the averaged data table in RRD.  Maybe we would need 102 data items, but the daily data only has 100 items. So rrdtool switches to the next longer period and uses the monthly data for the week table.

Or something similar.  The earliest required data just goes beyond the data in the table?

I have a hunch that we should either have a few items more in the RRD tables 
(currently 100, in practice there seems to be 101-120 items...)
```
root@router1:~# rrdtool info /tmp/rrd/router1/entropy/entropy.rrd
filename = "/tmp/rrd/router1/entropy/entropy.rrd"
rrd_version = "0001"
step = 30
last_update = 1589743306
ds[value].type = "GAUGE"
ds[value].minimal_heartbeat = 60
ds[value].min = 0.0000000000e+00
ds[value].max = 4.2949672950e+09
ds[value].last_ds = "UNKN"
ds[value].value = 2.2544000000e+04
ds[value].unknown_sec = 0
rra[0].cf = "AVERAGE"
rra[0].rows = 120
rra[0].pdp_per_row = 1
rra[0].xff = 1.0000000000e-01
rra[0].cdp_prep[0].value = NaN
rra[0].cdp_prep[0].unknown_datapoints = 0
rra[1].cf = "AVERAGE"
rra[1].rows = 103
rra[1].pdp_per_row = 28
rra[1].xff = 1.0000000000e-01
rra[1].cdp_prep[0].value = 2.0634933333e+04
rra[1].cdp_prep[0].unknown_datapoints = 0
rra[2].cf = "AVERAGE"
rra[2].rows = 101
rra[2].pdp_per_row = 201
rra[2].xff = 1.0000000000e-01
rra[2].cdp_prep[0].value = 5.4665333333e+03
rra[2].cdp_prep[0].unknown_datapoints = 0
rra[3].cf = "AVERAGE"
rra[3].rows = 101
rra[3].pdp_per_row = 892
rra[3].xff = 1.0000000000e-01
rra[3].cdp_prep[0].value = 5.4144620000e+05
rra[3].cdp_prep[0].unknown_datapoints = 0
rra[4].cf = "AVERAGE"
rra[4].rows = 101
rra[4].pdp_per_row = 10540
rra[4].xff = 1.0000000000e-01
rra[4].cdp_prep[0].value = 5.2331778667e+06
rra[4].cdp_prep[0].unknown_datapoints = 3357
```

hnyman, 2020-05-17:
Right now the graph suddenly switched to displaying longer period data:

Time is 22:27 and the oldest data in daily table is from yesterday 22:40 

[image]

Now time is 22:30 and graph is again ok:
[image]

A few minutes earlier there was still 22:26 data item from yesterday and the graph was ok, but once that disappeared, the graph got wrong for 1-2 minutes (wrong maybe between 22:28-22:29).

```
                <database>
                        <!-- 2020-05-16 22:26:00 EEST / 1589657160 --> <row><v> 2.0638047619e+03 </v></row>
                        <!-- 2020-05-16 22:40:00 EEST / 1589658000 --> <row><v> 1.9962452381e+03 </v></row>
```

After update
```
                <database>
                        <!-- 2020-05-16 22:40:00 EEST / 1589658000 --> <row><v> 1.9962452381e+03 </v></row>
                        <!-- 2020-05-16 22:54:00 EEST / 1589658840 --> <row><v> 1.8190857143e+03 </v></row>
                        <!-- 2020-05-16 23:08:00 EEST / 1589659680 --> <row><v> 1.6542809524e+03 </v></row>
```

My guess is that the rough data granularity set by us, only 100 data items, plays wrong for some short periods when the nearest data point of the required start time gets calculated.  collectd seems to create 120 point for the hourly data, as that is the minimum for 30sec intervals. But for longer periods it seems to be pretty much the required 100 plus 1 item. (but for daily data 103 for some reason).

Might be that the error has been there also earlier, but we have just noticed it now.

In any case, the error is likely reproduceable by changing the required time period.

jow-, 2020-05-17:
So maybe we should simply increase the amount of data points? The original values are extremely conservative and date back over ten years, when this thing was supposed to run on wrt54gl style devices...

hnyman, 2020-05-17:
I will play with this a bit and try to figure out a bit more.
But your idea might be the right one.

Other option might be to force 1-2 more data points to be created by collectd.

Schneehuhn, 2020-05-17:
> Might be that the error has been there also earlier, but we have just noticed it now.

I am on 19.07 (footer of webpage says `Powered by LuCI openwrt-19.07 branch (git-20.057.55219-13dd17f) / OpenWrt 19.07.2 r10947-65030d81f3`), so likely the bug wasn't introduced with the move to JS.

hnyman, 2020-05-17:
Summary:

**The bug/feature is likely a combination of data sets from different periods.
I got the wrong graphs on times when the newest weekly datapoint had been stored after the last daily datapoint.
On those times, the longer period dataset apparently provided a better match for the time period asked by LuCI**

Mitigation proposal:

**change the default RRARows 100 to 144, which seems to lead to nice even averaged data RRA periods:
(30 sec, 10 min, 70 min, 5h10min, 2d13h)**

----

Teh bug was a bit harder to reproduce than I thought.

I managed to get faulty graphs again at midnight, but I failed at a few previous database change points (and I did not get faulty graphs at the next interval after the faulty one...)

This time the faulty period was for a few minutes just before the new data update (while the time was after the stored time from yesterday, but before the next update point today).

```
root@router1:~# date; rrdtool dump /tmp/rrd/router1/entropy/entropy.rrd  | head -n 162 | tail -n 3 ; echo "..."; rrd
tool dump /tmp/rrd/router1/entropy/entropy.rrd  | head -n 264 | tail -n 3
Mon May 18 00:04:05 EEST 2020
                <database>
                        <!-- 2020-05-17 00:04:00 EEST / 1589663040 --> <row><v> 2.0779928571e+03 </v></row>
                        <!-- 2020-05-17 00:18:00 EEST / 1589663880 --> <row><v> 1.8566261905e+03 </v></row>
...
                        <!-- 2020-05-17 23:38:00 EEST / 1589747880 --> <row><v> 1.4186928571e+03 </v></row>
                        <!-- 2020-05-17 23:52:00 EEST / 1589748720 --> <row><v> 1.3693309524e+03 </v></row>
                </database>
root@router1:~# echo wrong
wrong

root@router1:~# date; rrdtool dump /tmp/rrd/router1/entropy/entropy.rrd  | head -n 162 | tail -n 3 ; echo "..."; rrd
tool dump /tmp/rrd/router1/entropy/entropy.rrd  | head -n 264 | tail -n 3
Mon May 18 00:05:57 EEST 2020
                <database>
                        <!-- 2020-05-17 00:04:00 EEST / 1589663040 --> <row><v> 2.0779928571e+03 </v></row>
                        <!-- 2020-05-17 00:18:00 EEST / 1589663880 --> <row><v> 1.8566261905e+03 </v></row>
...
                        <!-- 2020-05-17 23:38:00 EEST / 1589747880 --> <row><v> 1.4186928571e+03 </v></row>
                        <!-- 2020-05-17 23:52:00 EEST / 1589748720 --> <row><v> 1.3693309524e+03 </v></row>
                </database>
root@router1:~# echo wrong
wrong
root@router1:~# date; rrdtool dump /tmp/rrd/router1/entropy/entropy.rrd  | head -n 162 | tail -n 3 ; echo "..."; rrd
tool dump /tmp/rrd/router1/entropy/entropy.rrd  | head -n 264 | tail -n 3
Mon May 18 00:06:20 EEST 2020
                <database>
                        <!-- 2020-05-17 00:18:00 EEST / 1589663880 --> <row><v> 1.8566261905e+03 </v></row>
                        <!-- 2020-05-17 00:32:00 EEST / 1589664720 --> <row><v> 1.6983690476e+03 </v></row>
...
                        <!-- 2020-05-17 23:52:00 EEST / 1589748720 --> <row><v> 1.3693309524e+03 </v></row>
                        <!-- 2020-05-18 00:06:00 EEST / 1589749560 --> <row><v> 1.3437261905e+03 </v></row>
                </database>
root@router1:~# echo ok
ok
```

Looking at the rrdtool sources, there is a determination logic for selecting the rrd period table to fetch data. The logic is in rrd_fetch.c roughly at lines 170-220. It aims for covering the whole data period and if that is not possible, the next best partial data set.  It uses the time step size several times in defining the "realistic period & step" vs. the original wishes.

The bug is likely a combination of data sets from different periods.
I think that I got the wrong graphs on times when the weekly datapoint had been stored after the last daily datapoint.

Last weekly datapoints:
```
                        <!-- 2020-05-17 22:19:30 EEST / 1589743170 --> <row><v> 1.3326341625e+03 </v></row>
                        <!-- 2020-05-18 00:00:00 EEST / 1589749200 --> <row><v> 1.4034275290e+03 </v></row>
```

[... 30 lines cut ...]
rra[1].cf = "AVERAGE"
rra[1].rows = 144
rra[1].pdp_per_row = 20
rra[1].xff = 1.0000000000e-01
rra[1].cdp_prep[0].value = 2.5899000000e+03
rra[1].cdp_prep[0].unknown_datapoints = 0
rra[2].cf = "AVERAGE"
rra[2].rows = 144
rra[2].pdp_per_row = 140
rra[2].xff = 1.0000000000e-01
rra[2].cdp_prep[0].value = 6.3229000000e+03
rra[2].cdp_prep[0].unknown_datapoints = 97
rra[3].cf = "AVERAGE"
rra[3].rows = 144
rra[3].pdp_per_row = 620
rra[3].xff = 1.0000000000e-01
rra[3].cdp_prep[0].value = 6.3229000000e+03
rra[3].cdp_prep[0].unknown_datapoints = 377
rra[4].cf = "AVERAGE"
rra[4].rows = 144
rra[4].pdp_per_row = 7320
rra[4].xff = 1.0000000000e-01
rra[4].cdp_prep[0].value = 6.3229000000e+03
rra[4].cdp_prep[0].unknown_datapoints = 2297
```

hnyman, 2020-05-18:
I changed the default from 100 to 144 with 591b918e.
Hopefully that is enough as the first fix.

If there is no negative feedback, I will backport that to 19.07 at some point.

Tackling the problem fully would require shortening the requested time period a bit more, in a scaled way, but I think that it is an overkill for a cosmetic problem that manifests itself only rarely.

closed 2020-05-18 by hnyman

hnyman, 2021-11-28:
@jow- 

Returning to the amount to data points in the LuCI statistics RRD databases.
I increased the default number from 100 to 144 in early 2020.
100 was really extremely conservative, like you said, and the current 144 is still really minimized.

That produces a quite roughly averaged graphs (especially with the daily and weekly periods).

Now I am considering to double the amount to 288, which would again produce nice database summary periods with our current duration definitions.

288 with our current durations `(1h,1d,1w,1m,1y)` would 
generate RRD with `30s, 5min, 35min, 2h35min, 1d6h30min` averaging periods
instead of the current `30s, 10min, 70min, 5h10min, 2d13h`.

I have been testing it in one of my routers for a while, and especially the daily and weekly periods look much more detailed.

Any objections for that change?
It would double the size of the RRD database (in RAMdisk), but that should not cause any trouble for modern devices.

jow-, 2021-11-28:
Fine with me, please go ahead!