openwrt/luci #4067: OpenWrt 19.07.3 | Mount Points missing from System menu in LuCI

ioanpintican, 2020-05-18:
OpenWrt 19.07.3 r11063-85e04e9f46 / LuCI openwrt-19.07 branch git-20.138.63234-ccd9d67 on Linksys WRT32X, block-mount (2020-05-12-84269037-1) installed and **Mount Points submenu is missing from System menu in LuCI**.

[image]

jow-, 2020-05-18:
Try to log out and in again, see if it appears then.

ioanpintican, 2020-05-18:
@jow- already tried that (I know that it takes some moving around the menus in order to make new ones appear), but it's not there...

jow-, 2020-05-18:
So you actually logged out and logged in again by entering your password to obtain a new session id?

ioanpintican, 2020-05-18:
Yep. Even tried different browsers & clear cache between sessions.
And I rebooted also.

jow-, 2020-05-18:
Ok, looks broken then.

ioanpintican, 2020-05-18:
Mounted the usb hdd form ssh session tho... :)
[image]

pesa1234, 2020-05-18:
Same also for me on 19.07.3

jow-, 2020-05-19:
Works for me after a device reboot.

ioanpintican, 2020-05-19:
Performed reset to defaults & flashed factory firmware.
Now it works for me too.
Thanks for the response :)

closed 2020-05-19 by ioanpintican

XinArkh, 2022-04-07:
You need to install `block-mount` package first.
```bash
opkg install block-mount
```
ref: https://forum.openwrt.org/t/no-mount-points-menu-solved/77907