openwrt/luci #4068: AVM 3000 - Switch switch0 has an unknown topology - VLANs > 15

christianbur, 2020-05-18:
I use an AVM FRITZ! repeater 3000 with OpenWrt 19.07.3 (r11063-85e04e9f46 / LuCI openwrt-19.07 branch git-20.136.49537-fb2f363), with version 19.07.2 the behaviour is the same. 

The CPU should support 128 VLANs:

```
root@avm3000:~# swconfig dev switch0 help
switch0: 90000.mdio(QCA AR40xx), ports: 6 (cpu @ 0), vlans: 128
```

I also use VLAN 20-23 for a while (created via /etc/config/network), with no problems .
But via the LuCI WebGui I always get the two error messages
`"Switch switch0 has an unknown topology - the VLAN settings might not be accurate"` under network->switch and in each VLAN line `"Expecting: value between 0 and 15"`.
Because of the two errors you cannot save `"Cannot save due to invalid values"`.

How can you tell LuCI that the internal switch supports 128 VLANs and not just 16?

[Screenshot](https://forum.openwrt.org/uploads/default/original/3X/4/a/4ac42b602523dc968b4f7fc935561b7c4cf2b1bc.png)

```
root@avm3000:~# cat /etc/config/network

config interface 'loopback'
	option ifname 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option ula_prefix 'fdxx:xxxx:xxxx::/48'

config interface 'lan'
	option type 'bridge'
	option proto 'dhcp'
	option ifname 'eth0'

config switch
	option name 'switch0'
	option reset '1'
	option enable_vlan '1'

config switch_vlan
	option device 'switch0'
	option vlan '20'
	option ports '0t 5t'

config switch_vlan
	option device 'switch0'
	option vlan '21'
	option ports '0t 5t'

config switch_vlan
	option device 'switch0'
	option vlan '22'
	option ports '0t 5t'

config switch_vlan
	option device 'switch0'
	option vlan '23'
	option ports '0t 5t'

config interface 'vlan21'
	option ifname 'eth1.21'
	option type 'bridge'
	option proto 'none'

config interface 'vlan22'
	option proto 'dhcp'
	option ifname 'eth1.22'
	option type 'bridge'
	option defaultroute '0'

config interface 'vlan23'
	option ifname 'eth1.23'
	option type 'bridge'
	option proto 'dhcp'
	option defaultroute '0'

config interface 'vlan20'
	option proto 'dhcp'
	option type 'bridge'
	option ifname 'eth1.20'
```

jow-, 2020-05-19:
Needs to be fixed in OpenWrt.

closed 2020-05-19 by jow-

christianbur, 2020-05-19:
how does luci determine the range 0-15?

jow-, 2020-05-19:
It's hardcoded fallbacks from here: https://github.com/openwrt/luci/blob/openwrt-19.07/modules/luci-mod-network/htdocs/luci-static/resources/view/network/switch.js#L149

LuCI uses the information from `ubus call luci getSwconfigFeatures '{ "switch": "switch0" }'` to discover the switch constraints and the contents of `/etc/board.json` to learn what switches are available.