openwrt/luci #4071: Show everything or show nothing

castillofrancodamian, 2020-05-19:
I see that some do not mind seeing the LuCI menus (not me either) when they are not logged in, but it does not show everything.
[image]

castillofrancodamian, 2020-05-21:
[image]

ptpt52, 2020-06-11:
this patch may fix this issue
```
diff --git a/applications/luci-app-wireguard/root/usr/share/luci/menu.d/luci-app-wireguard.json b/applications/luci-app-wireguard/root/usr/share/luci/menu.d/luci-app-wireguard
index 0cb87f3e2..a42677e1a 100644
--- a/applications/luci-app-wireguard/root/usr/share/luci/menu.d/luci-app-wireguard.json
+++ b/applications/luci-app-wireguard/root/usr/share/luci/menu.d/luci-app-wireguard.json
@@ -5,6 +5,9 @@
                "action": {
                        "type": "template",
                        "path": "wireguard"
+               },
+               "depends": {
+                       "acl": [ "luci-mod-status-index" ]
                }
        }
 }
```

castillofrancodamian, 2020-06-11:
Did you make a Pull Request?

closed 2020-07-04 by jow-

castillofrancodamian, 2020-07-04:
👍👍👍

PR #4221 (2020-07-04): luci-app-wireguard: fix acl issue
fixes #4071