openwrt/luci #4074: weblate: "Source in review" blocks translation

aparcar, 2020-05-19:
Moved from https://github.com/openwrt/luci/issues/3183#issuecomment-629805047

@jow- @aparcar 

I noticed that some apps, like luci-app-opkg, are partially closed for translation: most of their source strings are marked with "source in review" and the target language strings can't be edited. 

[image]

I have traced this to the fact that some apps contain unnecessary "English" translations, where one or more strings (or comments in weblate?) have been entered, and then weblate automatically thinks that the English translated string file should be used as the "source" instead of the msgid strings in the language .po. And as the English .po file is mostly empty, weblate declares all not-found source strings in other .po files as suspicious and refuses to allow translation for them.

[image]

Weblate also complains about the duplicate languages frequently after each refresh from LuCI github:

[image]

I tried to reset opkg's English translation (by clearing upstream the one "translated" string), but that did not help. Weblate still allows only a few strings to be translated, while all others are locked due to "source in review". Apparently it remembers that opkg has been "translated" to English at some point.

I see three different solutions:

1.  We remove all "en" English translations from LuCI. There is no real reason to have them as the original source code is in English.  
     * We might need to revamp the translation selection logic in LuCI, so that English would not be really there, or it would be like "English (no translation)". I tried to look at the LuCI source, and I do not see any real need for English, as the default setting is empty in config file, and the selection defaults to "auto" if there is no setting.  ( I wonder why we at all do create the empty i18n-xxx-en files ??? Is there some historical reason for that, jow ?)

2. We duplicate all strings in "en" English .po, just for weblate purposes. But that would be an additional step for the i18n-sync or similar tool.

3. We somehow manage to make weblate to forget about English translation. But so far I have been unable to find options for that.

aparcar, 2020-05-19:
Should we just symlink all `.pot` templates to `en.po`?

jow-, 2020-05-19:
Sounds like a weblate bug to me. I do not see why translating msgids to english would suddenly make them unavailable. I'm fine with committing changes that empty out english `.po` files but I don't want symlink trickery.

The reason why there's english `.po` files is that I do not want any special casing. It is just another language and I do want to reserve the possibility to override english source strings with another english translation (e.g. to introduce different punctuation, spelling etc.)

hnyman, 2020-05-19:
Like I said above, I tried that "committing changes that empty out english .po files", but that did not help. opkg translation is still semi.locked

I think that the underlying reason is that we declare in weblate settings that "English" is our source language, and then we have a mostly empty English translation separately from the actual source strings, so Weblate thinks that we want it to use that empty translation. (Thatswhy we get also get that weblate warning about duplicate language en,en).

One solution might be to change the "source" language to something non-usable, so that English would be just a translation in weblate. Each .po includes the source strings in any case. E.g. lets pick "Northern Sami" as the source, (the Sami dialect spoken by maybe 1000 people in Northern Norway & Finland). Not sure how that would play out, but might be worthwhile to test.  Sadly there is no "source code" offered as the source language option

https://hosted.weblate.org/settings/openwrt/#workflow 
[image]

Other part is that out settings for the components do not actually define the templates/xxx.pot as the source, but we say that en/xxx.po is the template for new languages for a component. That might be automatic due to the "English as source language", but it might be worthwhile to test changing that at least for some "components". But that would likely not change the existing problems, as 

https://hosted.weblate.org/settings/openwrt/luciapplicationscommands/#files 
[image]

Might be that we should open a ticket at weblate and ask for advice. Possibly we are overlooking something.

hnyman, 2020-05-20:
I found relevant discussion in the weblate bug tracker: https://github.com/WeblateOrg/weblate/issues/3471

Especially this: https://github.com/WeblateOrg/weblate/issues/3471#issuecomment-629215709

The proposed solution seems to be to define an imaginary language as the official source, and then English would lose its special status. A bit like my proposal of Northern Sami above.

jow-, 2020-05-20:
Then please go for it. The current Weblate interpretation of the gettext workflow is ... curious. To me it makes no sense at all declaring non -pot files as canonical translation source.

hnyman, 2020-05-20:
I feld a question to that thread, and asked for their advice (either for adding a language or modifying the lanague regex).
https://github.com/WeblateOrg/weblate/issues/3471#issuecomment-631260122

I tried to explain our situation pretty closely, so hopefully the possible answers could be to the point.

(Sofar I have not found any place to create that imaginary language)

hnyman, 2020-05-20:
> To me it makes no sense at all declaring non -pot files as canonical translation source.

Regarding that, the automatics may be partially causing the problems.

In our component settings (for each component) the default template for newly found components is declared to be the "en" file instead of the templatates/xxx.pot file. That is likely due to automatics based on English as source. 
(Example: https://hosted.weblate.org/settings/openwrt/luciapplicationsahcp/#files )

I have manually corrected e.g. opkg for the .pot, but that does not help. 

I wait a bit for their possible answer and will then experiment with the source language change.

hnyman, 2020-05-20:
My question at weblate bug tracker triggered some new commits from them, which has helped a lot.

They created "English (devel)" and clarified some errors/comments/advice.

After I changed the source language, I received lots of errors and crashes, and many strings were still locked. After I mentioned that in upstream, they added commits to prevent caching English status and to prevent crashes on source language changes. That seems to have cleared most problems.

FYI, weblate key commits are on two pages from here:
https://github.com/WeblateOrg/weblate/commits/de13c3dcf0207c7724cb1c7a387dd9f43c6312dd

I still haven't got any answer on how we can prevent casual users from translating "English (developer)", but the original main problem seems to have been mainly solved. I have tested with Finnish, and the previously locked strings seem now open.

Ps. the last git merge in LuCI repo caused a reloading of translations strings at weblate taking quite some time. Hopefully the next updates do not take an hour, and that was just one-time workload after the language change.

jow-, 2020-07-04:
Closing this as it appears to be addressed.

closed 2020-07-04 by jow-