openwrt/luci #4077: 19.07.3 regression: installing an app doesn't show in the menu unless logging out/in again

jonglezb, 2020-05-20:
In 19.07.2, after installing a LuCI app, it started appearing in the menu as soon as you go to a new page.

In 19.07.3, the new app doesn't appear in the menu even after changing page. A logout and login is needed.

I tried the bootstrap theme and the following apps:

- luci-app-openvpn
- luci-app-nlbwmon

To reproduce:

- log in
- go to System / Software
- update lists
- install `luci-app-nlbwmon`
- go to (for instance) Status / Overview
- the "Bandwidth Monitor" menu still does not appear
- log out / log in : the menu now appears as expected

jow-, 2020-05-20:
Yep, its a known issue which we cannot fix for 19.07.3 anymore. I added a note to the release notes. A fix will entail an update to luci-app-opkg which will flush the SessionStorage menu cache after a package install operation.

singalen, 2020-05-23:
Is there a workaround? I wouldn't like to reboot.
I tried 
```
/etc/init.d/uhttpd stop
rm -rf /tmp/luci-*
/etc/init.d/uhttpd start
```
but it didn't help.

zorun, 2020-07-06:
Would it be enough to backport the (secondary) fix from dc57e4bc6c83d6c869c318256057e3636bfc5d66 ?

jow-, 2020-07-17:
@zorun - yes, most likely. Not entirely sure if we'd need further backports in the dispatcher code wrt. server side cache invalidation

zorun, 2020-08-30:
It turns out that simply backporting the fix is enough, see #4402

zorun, 2020-09-03:
The fix is merged, thanks jow!

PR #4402 (2020-09-03): [19.07] luci-app-opkg: flush menu cache after opkg actions
This is a partial backport of dc57e4bc6c83d6c869c318256057e3636bfc5d66

Fixes: #4077
commit 0d0ab01a (2020-09-03): luci-app-opkg: flush menu cache after opkg actions

This is a partial backport of dc57e4bc6c83d6c869c318256057e3636bfc5d66

Fixes: #4077
closed 2020-09-03 by jonglezb