openwrt/luci #4082: Question: huge error log for Zerotier One

nemo2018, 2020-05-21:
Hi there,
not sure this is right place, I would just ask for some help.
My Luci version:
LuCI Master (git-20.117.60969-420c61a) / OpenWrt R20.4.8

running Zerotier version: 1.4.6.

Basically Zerotier is running very well, without anyproblem.

My question is at the system log of Luci, there are huge error log related to Zerotier, every 5 minutes. all system log are full of this. What should I check for it?
[image]

Thank you!

hnyman, 2020-05-21:
What is OpenWrt R20.4.8 ?

But that zerotier log output is not a GUI bug, so it does not belong to this LuCI GUI bug tracker.

zerotier is in the packages repo, so you should file the bug there (https://github.com/openwrt/packages/issues )  and maybe notify the package maintainer @mwarning

closed 2020-05-21 by hnyman

mwarning, 2020-05-21:
I suspect the miniupnpc component of zerotier.