openwrt/luci #4083: Wi-Fi buttons do weird things

castillofrancodamian, 2020-05-21:
It seems to me to see that if I enable or disable one Wi-Fi interface, the other one on the other radio is also disabled.

[image]
[image]

castillofrancodamian, 2020-05-22:
In the wireless configuration file I see that the _disabled_ option is at 0 of the two radios, but one is working.

jow-, 2020-05-24:
That does not sound like a LuCI bug to me, probably an issue in the underlying OpenWrt subsystem

jow-, 2020-05-29:
Seems to be a bug in OpenWrt: https://forum.openwrt.org/t/wrt1900acsv2-only-one-wifi-radio-is-staying-up/64273

closed 2020-05-29 by jow-