openwrt/luci #4089: getHostHints may send local IP addresses to upstream DNS server

janh, 2020-05-22:
The ubus method `getHostHints` sends PTR queries for IP addresses in the local network to to the locally configured resolver (which depending on configuration could be an external DNS server). This can leak information that would otherwise not be visible externally, including RFC1918 addresses, as well as ULAs and link-local IPv6 addresses.

In the default configuration of OpenWrt this is a lesser issue, as dnsmasq is used as local resolver and reverse lookups for private addresses are not forwarded due to the `boguspriv` option. But even then, queries for public IPv6 addresses in the local network are forwarded to the upstream resolver.

jow-, 2020-05-22:
So, what solution do you suggest?

janh, 2020-05-23:
I am not quite sure how the solution should look like. My first idea was to only perform RDNS lookups for host hints when the resolver is local to the device or at least in the same subnet.

But I'm actually wondering whether doing RDNS lookups is really necessary. The same information should be available from the DHCP leases. However, that only works when the DHCP server is on the same device.

So the question is whether `getHostHints` is supposed to provide hostnames in situations where the DHCP server is on another device (e.g. when running on an access point). In this case RDNS lookups make sense, but right now they don't work anyway in the default configuration (as the local dnsmasq wont't forward the queries). And the configuration that makes them work is the same one that allows the information leak.