openwrt/luci #4090: The 'startup' page using the OpenWRT Luci theme, results in "XHR request timed out"

jeclark2006, 2020-05-22:
I could not find a report of a similar problem as the subject.

Basically what is the System -> Startup page supposed to do, and what would contribute to the indicated error, "XHR request timed out", and a continual 'loading...' message.

Thanks
John Clark.

castillofrancodamian, 2020-05-22:
Here's something like this is when you manually stop loading the page

https://user-images.githubusercontent.com/47670442/76365543-46fa8c00-6306-11ea-8719-3503d2a95603.png

jow-, 2020-05-22:
A broken init script could cause it. Please provide the current output of `ls -lh /etc/init.d/`

jeclark2006, 2020-05-22:
What is meant by 'a broken init script'?

Here's the output of the listing on /etc/init.d

root@RaptorXR-10:/www#  ls -lh /etc/init.d/
-rwxr-xr-x    1 root     root          64 Nov 22 17:53 WSRadioDown
-rwxr-xr-x    1 root     root         692 Nov 22 17:53 atd
-rwxr-xr-x    1 root     root        1.4K Nov 22 17:53 boot
-rwxr-xr-x    1 root     root        7.9K Nov 22 17:53 collectd
-rwxr-xr-x    1 root     root         324 Nov 22 17:53 conntrackd
-rwxr-xr-x    1 root     root         821 Nov 22 17:53 cron
-rwxr-xr-x    1 root     root         367 Nov 22 17:53 ddns
-rwxr-xr-x    1 root     root         340 Nov 22 17:53 diagBoot
-rwxr-xr-x    1 root     root       29.4K Nov 22 17:53 dnsmasq
-rwxr-xr-x    1 root     root         255 Nov 22 17:53 done
-rwxr-xr-x    1 root     root        5.8K Nov 22 17:53 dropbear
-rwxr-xr-x    1 root     root         997 Nov 22 17:53 firewall
-rwxr-xr-x    1 root     root        5.6K Nov 22 17:53 fwknopd
-rwxr-xr-x    1 root     root        1013 Nov 22 17:53 gpio_switch
-rwxr-xr-x    1 root     root         841 Nov 22 17:53 gpsd
-rwxr-xr-x    1 root     root        3.4K Nov 22 17:53 led
-rwxr-xr-x    1 root     root         685 Nov 22 17:53 lighttpd
-rwxr-xr-x    1 root     root         214 Nov 22 17:53 lm-sensors
-rwxr-xr-x    1 root     root        2.3K Nov 22 17:53 log
-rwxr-xr-x    1 root     root         652 Nov 22 17:53 luci_statistics
-rwxr-xr-x    1 root     root         323 Nov 22 17:53 lvm2
-rwxr-xr-x    1 root     root        2.6K Nov 22 17:53 network
-rwxr-xr-x    1 root     root         292 Nov 22 17:53 odhcpd
-rwxr-xr-x    1 root     root         391 Nov 22 17:53 qos
-rwxr-xr-x    1 root     root        2.3K Nov 22 17:53 relayd
-rwxr-xr-x    1 root     root         368 Nov 22 17:53 rpcd
-rwxr-xr-x    1 root     root         232 Nov 22 17:53 rsyncd
-rwxr-xr-x    1 root     root        9.5K Nov 22 17:53 snmpd
-rwxr-xr-x    1 root     root         749 Nov 22 17:53 snort
-rwxr-xr-x    1 root     root        8.2K Nov 22 17:53 stunnel
-rwxr-xr-x    1 root     root        1.2K Nov 22 17:53 sysctl
-rwxr-xr-x    1 root     root         662 Nov 22 17:53 sysfixtime
-rwxr-xr-x    1 root     root        1.2K Nov 22 17:53 sysfsutils
-rwxr-xr-x    1 root     root         248 Nov 22 17:53 syslog-ng
-rwxr-xr-x    1 root     root        2.1K Nov 22 17:53 sysntpd
-rwxr-xr-x    1 root     root         986 Nov 22 17:53 system
-rwxr-xr-x    1 root     root        1.3K Nov 22 17:53 ucitrack
-rwxr-xr-x    1 root     root        5.3K Nov 22 17:53 uhttpd
-rwxr-xr-x    1 root     root         125 Nov 22 17:53 umount
-rwxr-xr-x    1 root     root         239 Nov 22 17:53 urandom_seed
-rwxr-xr-x    1 root     root         220 Nov 22 17:53 urngd
-rwxr-xr-x    1 root     root         195 Nov 22 17:53 vsftpd

jow-, 2020-05-22:
Whats happens if you invoke `ubus call luci getInitList` via SSH?

jeclark2006, 2020-05-22:
While it does take a little time to return with a list, it does return a list. The web page, just hangs.

John

---- output of root@RaptorXR-10:/www#  ubus call luci getInitList

root@RaptorXR-10:/www#  ubus call luci getInitList
{
	"stunnel": {
		"index": 90,
		"enabled": true
	},
	"gpio_switch": {
		"index": 94,
		"enabled": true
	},
	"odhcpd": {
		"index": 35,
		"enabled": true
	},
	"rpcd": {
		"index": 12,
		"enabled": true
	},
	"dropbear": {
		"index": 19,
		"enabled": true
	},
	"rsyncd": {
		"index": 90,
		"enabled": true
	},
	"led": {
		"index": 96,
		"enabled": true
	},
	"ucitrack": {
		"index": 80,
		"enabled": true
	},
	"cron": {
		"index": 50,
		"enabled": true
	},
	"fwknopd": {
		"index": 95,
		"enabled": true
	},
	"collectd": {
		"index": 80,
		"enabled": true
	},
	"snmpd": {
		"index": 50,
		"enabled": true
	},
	"uhttpd": {
		"index": 50,
		"enabled": true
	},
	"done": {
		"index": 95,
		"enabled": true
	},
	"system": {
		"index": 10,
		"enabled": true
	},
	"lm-sensors": {
		"index": 28,
		"enabled": true
	},
	"gpsd": {
		"index": 50,
		"enabled": true
	},
	"ddns": {
		"index": 95,
		"enabled": true
	},
	"luci_statistics": {
		"index": 79,
		"enabled": true
	},
	"qos": {
		"index": 50,
		"enabled": true
	},
	"WSRadioDown": {
		"index": 255,
		"enabled": false
	},
	"vsftpd": {
		"index": 50,
		"enabled": true
	},
	"lighttpd": {
		"index": 50,
		"enabled": true
	},
	"umount": {
		"index": 255,
		"enabled": false
	},
	"urngd": {
		"index": 0,
		"enabled": true
	},
	"sysfsutils": {
		"index": 11,
		"enabled": true
	},
	"sysntpd": {
		"index": 98,
		"enabled": true
	},
	"urandom_seed": {
		"index": 99,
		"enabled": true
	},
	"network": {
		"index": 20,
		"enabled": true
	},
	"atd": {
		"index": 50,
		"enabled": true
	},
	"conntrackd": {
		"index": 21,
		"enabled": true
	},
	"syslog-ng": {
		"index": 50,
		"enabled": true
	},
	"diagBoot": {
		"index": 255,
		"enabled": true
	},
	"log": {
		"index": 12,
		"enabled": true
	},
	"snort": {
		"index": 90,
		"enabled": true
	},
	"sysfixtime": {
		"index": 0,
		"enabled": true
	},
	"firewall": {
		"index": 19,
		"enabled": true
	},
	"sysctl": {
		"index": 11,
		"enabled": true
	},
	"boot": {
		"index": 10,
		"enabled": true
	},
	"dnsmasq": {
		"index": 19,
		"enabled": true
	},
	"relayd": {
		"index": 80,
		"enabled": true
	},
	"lvm2": {
		"index": 15,
		"enabled": true
	}
}

porroso, 2020-05-24:
try

chmod +x /usr/libexec/rpcd/luci.ddns
/etc/init.d/rpcd restart

jow-, 2020-05-24:
Can you check how long it takes to return that list? Retry with `time ubus call luci getInitList`

jeclark2006, 2020-05-25:
I'll not post the resulting JSON list... but here's the time results:

real	0m 22.89s
user	0m 0.00s
sys	0m 0.00s

So, about 23 seconds to return a response. Is there some timeout value that is set to low? And if so, how to change that?

jeclark2006, 2020-05-26:
I didn't notice this before. But it seems like the command

ubus call luci getInitList

Actually causes the init scripts to run. In particular in my rc.local I have commands to 'mount' a USB memory device as a file system structured device.

Hence the first few lines of the following output are from that process.

The question I have is, is it true that when the browser access the  the System -> startup page, it invokes the scripts associated with the init.d and rc.local?

---- output ----
root@RaptorXR:/# ubus call luci getInitList
Mounting /dev/sda1 /video/lib
[ 5444.427158] [EXFAT] trying to mount...
[ 5444.431930] [EXFAT] mounted successfully
Mounting /dev/sda1 /video/lib
mount: mounting /dev/sda1 on /video/lib failed: Resource busy
{
        "stunnel": {
                "index": 90,
                "enabled": true
        },
        "gpio_switch": {
                "index": 94,
                "enabled": true
        },
        "odhcpd": {
                "index": 35,
                "enabled": true
        },
        "rpcd": {
                "index": 12,
                "enabled": true
        },
        "dropbear": {
                "index": 19,
                "enabled": true
        },
        "rsyncd": {
                "index": 90,
                "enabled": true
        },
        "led": {
                "index": 96,
                "enabled": true
        },
        "ucitrack": {
                "index": 80,
                "enabled": true
        },
        "cron": {
                "index": 50,
                "enabled": true
        },
        "fwknopd": {
                "index": 95,
                "enabled": true
        },
        "collectd": {
                "index": 80,
                "enabled": true
        },
        "snmpd": {
                "index": 50,
                "enabled": true
        },
        "uhttpd": {
                "index": 50,
                "enabled": true
        },
        "done": {
                "index": 95,
                "enabled": true
        },
        "system": {
                "index": 10,
                "enabled": true
        },
        "lm-sensors": {
                "index": 28,
                "enabled": true
        },
        "gpsd": {
                "index": 50,
                "enabled": true
        },
        "ddns": {
                "index": 95,
                "enabled": true
        },
        "luci_statistics": {
                "index": 79,
                "enabled": true
        },
        "qos": {
                "index": 50,
                "enabled": true
        },
        "WSRadioDown": {
                "index": 255,
                "enabled": false
        },
        "vsftpd": {
                "index": 50,
                "enabled": true
        },
        "lighttpd": {
                "index": 50,
                "enabled": true
        },
        "umount": {
                "index": 255,
                "enabled": false
        },
        "urngd": {
                "index": 0,
                "enabled": true
        },
        "sysfsutils": {
                "index": 11,
                "enabled": true
        },
        "sysntpd": {
                "index": 98,
                "enabled": true
        },
        "urandom_seed": {
                "index": 99,
                "enabled": true
        },
        "network": {
                "index": 20,
                "enabled": true
        },
        "atd": {
                "index": 50,
                "enabled": true
        },
        "conntrackd": {
                "index": 21,
                "enabled": true
        },
        "syslog-ng": {
                "index": 50,
                "enabled": true
        },
        "diagBoot": {
                "index": 255,
                "enabled": true
        },
        "log": {
                "index": 12,
                "enabled": true
        },
        "snort": {
                "index": 90,
                "enabled": true
        },
        "sysfixtime": {
                "index": 0,
                "enabled": true
        },
        "firewall": {
                "index": 19,
                "enabled": true
        },
        "sysctl": {
                "index": 11,
                "enabled": true
        },
        "boot": {
                "index": 10,
                "enabled": true
        },
        "dnsmasq": {
                "index": 19,
                "enabled": true
        },
        "relayd": {
                "index": 80,
                "enabled": true
        },
        "lvm2": {
                "index": 15,
                "enabled": true
        }
}
root@RaptorXR:/#

jow-, 2020-05-26:
It sources each script in `/etc/init.d/` to extract `START` and `STOP` variables. Furthermore it invokes `/etc/init.d/xxx enabled` for each script to determine whether it is enabled.

Properly written init scripts should do nothing when sourced or called with `enabled` as argument. I suspect there's one or more custom init scripts (or maybe init scripts by packages) which behave differently, causing the long delay.

jeclark2006, 2020-05-29:
That seems to be it. I guess I had thought that the entries in the webpage would not have invoked the scripts in init.d/* themselves, but matched up the existence of an entry in rc.d/* with what was found in init.d/*.

So my custom scripts which have to be run before rc.local is, were just the shell commands required, and not put into such things as 'boot {}' etc...

jow-, 2020-07-04:
Closing this as it was caused by changes unrelated to LuCI.

closed 2020-07-04 by jow-

logon84, 2021-02-10:
Same issue here, but when running "ubus call luci getInitList" I get "Command failed: Request timed out". What can I check?

logon84, 2021-02-10:
> Same issue here, but when running "ubus call luci getInitList" I get "Command failed: Request timed out". What can I check?

solved resetting to factory settings and then restoring a backup

castillofrancodamian, 2021-02-10:
> > Same issue here, but when running "ubus call luci getInitList" I get "Command failed: Request timed out". What can I check?
> 
> solved resetting to factory settings and then restoring a backup

It appears to me from time to time.

DjDiabolik, 2021-10-30:
Hi boys... i have the same similar issue whit my router whit openwrt.

When try to access to GUI immediatelly after the credential i see this errors.... but the GUI it's be loaded normally after that errors it's appears on my screen of pc.

It's normal ?

JFtico, 2021-11-14:
TIP: if you get the timeout on the  "ubus call luci getInitList"

The problem is the presence of **/etc/init.d/dsl_control**  just nuke that and the list returns in a second or so.