openwrt/luci #4108: Cannot restore configuration on OpenWrt 19.07.3

wojciechczyz, 2020-05-28:
I am trying to restore the backup of configuration, performed minutes ago. It fails with:
The uploaded backup archive is not readable

I am uploading the same tar file that is getting downloaded. Tried it few times, restarting router and backing up configuration again, still getting the same problem.

Linksys WRT1900ACS quite clean setup.

What it might be? How can I troubleshoot this?

hnyman, 2020-05-28:
Does it work from SSH console?
If you upload the tar file to the router with scp and try restoring it with the sysupgrade script?

Does the tar.gz archive look normal from console? And in your PC? Can you manually see its contents? (either from command line, or with GUI tools like WinRAR)

Trying to figure if that is a LuCI bug, download bug, upload bug or a sysupgrade bug...

jow-, 2020-05-29:
Cannot reproduce this issue. LuCI invokes this command to check if the archive is readable: https://github.com/openwrt/luci/blob/master/modules/luci-mod-system/htdocs/luci-static/resources/view/system/flash.js#L111

If `tar -tzf` fails, the error is emitted. Maybe something is interfering with the file upload... a virus scanner addon or similar.

richardhd, 2020-06-08:
I have the same issue. I noticed that LuCl generates a tar archive, but sysupgrade generates a tar.gz file. If I compress the the LuCL generated tar file to get a tar.gz file, then the LuCL restore does not generate an error message and it seems to work.

jow-, 2020-07-04:
Closing this as worksforme. We really expect a .tar.gz here.

closed 2020-07-04 by jow-