openwrt/luci #4110: Luci user auth issues with Chromium-based Microsoft Edge

atiensivu, 2020-05-29:
This is a bit of a bizarre problem since it only seems to be showing up on recent trunk builds, and I'm not entirely sure when it "broke". 

I have a Netgear R7800 running:

LuCI Master (git-20.149.19986-80580d7) / OpenWrt SNAPSHOT r13388-e430376b48
openwrt-ipq806x-generic-netgear_r7800-squashfs-sysupgrade.bin

The failing browser:
Microsoft Edge Version 85.0.530.0 (Official build) canary (64-bit)

It works completely fine with a previous build, on a slightly different router (Archer C59 v2):
LuCI Master (git-20.147.57899-d2dfc52) / OpenWrt SNAPSHOT r13060-471b8bf8c1

So it looks like somewhere between the two, something changed that broke Edge.

hnyman, 2020-05-29:
HTTP or HTTPS ?

If HTTPS, sometimes browsers are too eager to cache stuff, so that  they fail to see that the SSL certs have changed, or something similar.

Ps. I just used Edge "83.0.478.37 (Official build) (64-bit)" to login into my R7800 running HTTPS with master OpenWrt SNAPSHOT r13373-e8fbb98c6d / LuCI Master git-20.145.56006-f6f4005

jow-, 2020-05-29:
Same here, unable to reproduce. Try removing the `sysauth` cookie and see if log ins work again.

closed 2020-05-29 by jow-