openwrt/luci #4121: A small spelling mistake in English (screen DHCP and DNS)

mackonsti, 2020-05-30:
Hello everyone, many warm thanks for your great work and continuous support.

I just installed (converted) OpenWrt **19.07.3** to my Asus RT-AC58U and discovered a small spelling mistake whilst browsing the menus.

In DNS and DHCP there is a tab showng "Resolv" instead of "Resolv**e** and Host Files".

[image]

You could perhaps consider writing "Lease file" instead of one single word, too.

Thank you again.

dibdot, 2020-05-31:
The speeling is OK, see https://en.wikipedia.org/wiki/Resolv.conf

closed 2020-05-31 by dibdot

mackonsti, 2020-05-31:
Thank you @dibdot sorry for missing that out.