openwrt/luci #4125: luci-mod-network: split dns and dhcp functionality

wt, 2020-06-01:
I would like to be able to split out the functionality for dns and dhcp for dnsmasq such that the dns UI could be independently installed. This would be useful in cases like Turris images that use a completely different dns resolver (knot-resolver).

I am happy to help in this effort. Is this something that would be welcome? I think the first obvious stel would be separating the `config domain` sections and any other dns configuration out of the /etc/config/dhcp so that they can be maintained as separate files. Of course, appropriate backward compatibility would need to be added to make this work smoothly. Is there a statement anywhere on backward compat requirements for OpenWrt that I could read up on?

Is this something that would be possible?

stangri, 2020-06-02:
> I would like to be able to split out the functionality for dns and dhcp for dnsmasq such that the dns UI could be independently installed. This would be useful in cases like Turris images that use a completely different dns resolver (knot-resolver).

If there was a companion luci app for `kresd`, it could have been independently installed as things are right now. That is the case with `unbound` and `luci-app-unbound`.

There very well may be the case for splitting dnsmasq config out of the `/etc/config/dhcp` file, but unless I'm misunderstood your issue, the web UI for other resolvers can be installed independently as things are.

wt, 2020-06-03:
It's not only a matter of being able to configure kresd. It's also matter
of being able to hide the ui for dnsmasq UI when it is not used as the new
resolver. Having config that mostly does nothing is not super intuitive.
Having said that, there are options that kresd reads from the main section
of the /etc/config/dhcp as well, like the local domain. I think it might be
reasonable to extract the common dns options into something that can more
readily be used by more than one dns resolver. It should of course be
extensible by whatever luci extensions are installed to manage such dns
resolvers. I don't have specific ideas on the specific refactors to code
that I would make to accomplish this. I basically want to just get support
for the idea of doing this. I think it could really change the game for how
alternative resolvers are handled while no regressing dnsmasq support.

jow-, 2020-06-03:
Most UIs related to dnsmasq should disappear if dnsmasq is not present.

wt, 2020-06-04:
To provide a bit now context, dnsmasq is still used for DHCP on turris. I
would like to make it so that dnsmasq's luci config for DHCP stays around.
Also, there are some options that probably make sense for most DNS servers.
It would be ideal to separate those out so that different resolvers could
use a single common config for those common things.

>

jow-, 2020-07-04:
This looks like something that should be implemented in OpenWrt first to sort out the general configuration issues. LuCI can follow it then. I am not going to introduce LuCI specific deviations here.

closed 2020-07-04 by jow-