openwrt/luci #4132: Luci marks all interfaces for deletion

pacman366, 2020-06-04:
After updating openwrt to 19.07.3 (luci 20.154.47914-8bf301a), luci marks all my interfaces for deletion upon clicking Network > Interfaces, preventing me from making edits to them. Prior to the upgrade, I was on openwrt 19.07.2 and was able to edit my interfaces normally. Let me know if there's anything I can provide to debug this.

[image]

jow-, 2020-06-04:
Cannot reproduce this here.

pacman366, 2020-06-04:
Never mind, I found an obscure typo in my /etc/config/network. removing the typo fixed the issues.

``

closed 2020-06-04 by pacman366

spapadim, 2020-06-13:
For anyone else looking (I was staring at my config for 30min to find the obvious, as is usually case :), in case it helps: my "obscure typo" was simply a forgotten closing quote in one of the option values.  

This led to Luci showing "marked for deletion" next to each interface in UI.  Furthermore, only interfaces up the the one with the missing quote were shown in the UI; interfaces after that in the config did not show up in UI (guess that should have been a hint about where to look for typo..)