openwrt/luci #4135: luci-app-advanced-reboot: Warning: Unable to obtain device information!

pesa1234, 2020-06-07:
Hi all,
I have luci-app-advanced-reboot git-20.157.47124-7b4a785 on WRT1900ACS with latest SNAPSHOT SNAPSHOT r13503-050c31fa26 and I get the error on title.
What I have do wrong?

Thanks
[image]

stangri, 2020-06-09:
Linksys WRT-series devices were recently renamed in master, I have fixed it in this version: https://github.com/stangri/openwrt-repo/blob/master/luci-app-advanced-reboot_git-20.155.55664-f35803e-54_all.ipk if you could test it and let me know I'd appreciate it.

pesa1234, 2020-06-09:
[image]
Works perfectly, sorry but I don't see that the name was changed otherwise I was more accurate on bug report.

Thanks

closed 2020-06-09 by pesa1234

1Jeff1, 2020-06-09:
Also fixed on wrt32x

[image]