openwrt/luci #4150: luci-app-ddns - using custom option fails

mavack, 2020-06-10:
Trying to do a custom ddns server entry.

1. When selecting new --Custom -- fill in all fields click save, the DDNS Service provider field defaults back to 3322.org (The first one on the list) (In r13247 it totally failed but looks like something changed slightly between versions)

2. Both Custom update-URL and Custom update-script are mandatory fields, they are actually OR fields. 
https://openwrt.org/docs/guide-user/base-system/ddns

Trying to fill anything into the update-script field when only needing url triggers a failure.
203203  WARN : Custom update_script not found! - TERMINATE
 203203  WARN : PID '10342' exit WITH ERROR '1' at 2020-06-10 20:32

manually updating the etc/config/ddns file and removing update-script resolves the ddns working 

Using this with cloudns.net which use's an update URL only.

LuCI Master (git-20.160.76921-16f443b) / OpenWrt SNAPSHOT r13527-61307544d1

mavack, 2020-06-13:
Confirmed #2 is fixed, however part #1 still remains. 

When you have custom ddns provider configured, and you edit the config, the form pre-fills with config like it should, however the "DDNS Service provider [IPv4]" field defaults to 3322.org.   

I would guess you are pulling service_name to populate this field but since custom ddns provider does not have a defined uci service_name it just defaults to the first on the list.  If no service_name is returned then the field should display custom.

Ansuel, 2020-06-13:
Will check again but I tested and the problem was fixed.

mavack, 2020-06-30:
Issue 1 is still not resolved, 
Create new Custom DDNS provider
Save the config
go back to the ddns screen
click edit
The displayed configuration now defaults to provider showing 3322.org (first in the list) you need to change it back to custom before save. 

luci-app-ddns - git-20.181.72449-1f89ee8 - LuCI Support for Dynamic DNS Client (ddns-scripts)

		o.cfgvalue = function(section_id) {
			return uci.get('ddns', section_id, 'service_name');
		}

A custom DNS provider does not have a defined service_name UCI attribute as such it cannot return/display "-- custom --"
		o.value('-',"-- " + _("custom") + " --");

which is manually added to the list 2 lines above. 

Need if uci.get return nothing then display custom. This may or may not work as not sure if there are other exceptions that could be broken by returning custom only when no service_name is found.