openwrt/luci #4159: ubus call luci-rpc getDHCPLeases: The same device returns different mac address (DHCP, DHCPv6) information

ZoZhang, 2020-06-11:
My router is only connected to one device. 
Why did I return 2 different mac addresses (v4 and v6) when getting the dhcp device information ?

Am I the only one experiencing this problem ?

````
root@OpenWrt:~# ubus call luci-rpc getDHCPLeases
{
	"dhcp_leases": [
		{
			"expires": 41004,
			"hostname": "local",
			"macaddr": "00:E0:4C:68:4A:2F",
			"duid": "01:00:e0:4c:68:4a:2f",
			"ipaddr": "192.168.1.145"
		}
	],
	"dhcp6_leases": [
		{
			"expires": 39596,
			"macaddr": "25:A3:30:4C:14:10",
			"duid": "0001000125a3304c14109fe10235",
			"ip6addr": "fd56:ece5:8cd8::79f",
			"ip6addrs": [
				"fd56:ece5:8cd8::79f"
			]
		}
	]
}
````

jow-, 2020-06-11:
The MAC of the DHCPv6 lease was extracted from the DUID. Many modern DHCPv6 clients randomize the DUID for privacy reasons.

closed 2020-06-11 by jow-

ZoZhang, 2020-06-11:
Thanks for your patience. @jow-