openwrt/luci #4160: Please add missing sysupgrade options

zorxd, 2020-06-11:
Hi

sysupgrade has some interesting options, including the following:

-k : include in backup a list of current installed packages
-u : skip from backup files that are equal to those in /rom

It would be nice if they were added to LuCI as well. Personally, I would always choose these two options when upgrading. If you think adding these options would add to much bloat to the GUI, I think at least -k could be considered safe to add by default, perhaps -u as well. If a user didn't touch a file, it's likely the default one will be good after the upgrade. Maybe -u could even help prevent bugs when keeping configuration files and upgrading to a new major release.

closed 2021-08-09 by hnyman

shoulders, 2021-11-18:
hi

for newbies can someone explain what the following settings do?

-k : include in backup a list of current installed packages
-u : skip from backup files that are equal to those in /rom

[image]

And just confirm that `Keep settings and retain the current configuration` just preserves the config settings during the upgrade but not any of the binaries/packages installed by users.

My internet searches just bring me back here and these options are not explained anywhere.

thanks

zorxd, 2021-11-18:
-k creates a file with all packages installed before the update which will be available in installed_packages.txt after the update. So no more "Oh, which package am I missing?" since yes, software update doesn't preserve installed packages.

-u saves some space by not including unmodified configuration files in the backup. It also has the benefit of getting new default options if they changed with the new update, instead of getting stuck with legacy configuration files forever.

PR #4754 (2021-08-09): luci-mod-system: improve sysupgrade page
- Add missing -k and -u option from sysupgrade page.
- Fix missing check for image verification exit code
- Provide the actual reason of the image verification

Fixes: #4160

@jow- 

Here the screen of the fixes for the theme and the new option. (in the current state, with the material theme, the long text are never wrapped and goes outside the modal...)

[image]
commit 0f4f6e89 (2021-08-09): luci-mod-system: improve sysupgrade page

- Add missing -k and -u option from sysupgrade page.
- Fix missing check for image verification exit code
- Provide the actual reason of the image verification

Fixes: #4160
commit 63898260 (2021-08-18): luci-mod-system: improve sysupgrade page

- Add missing -k and -u option from sysupgrade page.
- Fix missing check for image verification exit code
- Provide the actual reason of the image verification

Fixes: #4160

(cherry picked from commit 0f4f6e89f15aa1dac0bfd676cb8364ef0b6c6e1f)