openwrt/luci #4174: luci: option domain in unbound.conf is lost on "apply changes"

AndreaBorgia-Abo, 2020-06-14:
Foreword: originally reported under [packages](https://github.com/openwrt/packages/issues/12444) but moved here as [recommended.](https://github.com/openwrt/packages/issues/12444#issuecomment-643671131)

Environment: Fritzbox 4040, OpenWRT 19.07.3

Description:
1. configure unbound in parallel mode
2. insert [relevant "domain" keyword](https://github.com/openwrt/packages/commit/b6caafef46829922dade557d6b3082360053bf8f) for custom domain in /etc/config/unbound
`option domain 'yourdomain'`
3. in the GUI make a minor change, say TTL from 120 to 121, and apply changes
4. verify that the keyword domain has disappeared from the config
5. again in the GUI make another minor chage, say TTL back to 120, and apply changes
6. verify that resolv.conf now has a "search lan." line instead of "search yourdomain."

Basically, after two restarts, the keyword is gone from resolv.conf
Issue first discussed in the [forum](https://forum.openwrt.org/t/unbound-option-domain-is-lost/64987/12?u=aboaboit)

jow-, 2020-06-14:
@EricLuehrsen

jow-, 2020-06-14:
@AndreaBorgia-Abo - the LuCI page was written in a way that the domain options depends on either `option dhcp_link none` or `option dhcp_link odhcpd`, so make sure that this option is set as well.

AndreaBorgia-Abo, 2020-06-14:
@jow- I have a use case for having "domain" also with "option dhcp_link dnsmasq". Would it be possible to rework the page in such a way that the keyword is kept in all cases? I have a workaround in place, for the moment.

jow-, 2020-06-14:
I guess so but I don't have any experience with unbound, nor am I maintaining it. Maybe @EricLuehrsen can chime in

EricLuehrsen, 2020-06-27:
I am working on this, but first I need to get the Unbound packaging itself squared away. There have been some issues.

closed 2020-07-17 by jow-

AndreaBorgia-Abo, 2020-07-26:
@EricLuehrsen I have just upgraded the unbound packages to these:
libunbound-heavy - 1.10.1-3
luci-app-unbound - git-20.204.30585-8750495-1
unbound-anchor - 1.10.1-3
unbound-control - 1.10.1-3
unbound-daemon-heavy - 1.10.1-3

Then I verified that my workaround in /usr/lib/unbound/unbound.sh was gone (as expected) and applied again [the steps above.](https://github.com/openwrt/luci/issues/4174#issue-638341342)
The issue is still there, so the first question I have is indeed: did I use the right version?

EricLuehrsen, 2020-07-26:
I have removed heavy and light prebuilt packages in master and replaced them with a default plus selected option build. So I guess the report is release 19.07. Changes needs to settle for time before I back port them.

AndreaBorgia-Abo, 2020-12-21:
Upgraded to 19.07.5, still with the same "heavy" package which means no backport yet, right?
I'm asking because apparently now "domain" survives a GUI edit even without a patch in /usr/lib/unbound/unbound.sh

AndreaBorgia-Abo, 2021-01-22:
Confirmed working also with 19.07.6, thank you.

PR #4270 (2020-07-17): luci-app-unbound: add rate_limit and dns_assist option support
maintainer: me
tested: Linksys WRT3200ACM
description:
resolves #4174 and picks up options from https://github.com/openwrt/packages/pull/12798