openwrt/luci #4182: Not found luci openwrt 19.07

predators46, 2020-06-16:
Not found luci openwrt 19.07

`  * opkg_download: Failed to download http://downloads.openwrt.org/releases/19.07.1/packages/mipsel_24kc/luci/luci_git-20.166.52937-b894a3b-1_all.ipk, wget returned 8.
 * opkg_install_pkg: Failed to download luci. Perhaps you need to run 'opkg update'?
 * opkg_install_cmd: Cannot install package luci.
Makefile:153: recipe for target 'package_install' failed
make[2]: *** [package_install] Error 255
Makefile:112: recipe for target '_call_image' failed
make[1]: *** [_call_image] Error 2
Makefile:196: recipe for target 'image' failed
make: *** [image] Error 2    `

hnyman, 2020-06-17:
You need to update your feeds.

There is now luci_git-20.167.61968-87da00a-1_all.ipk

(Buildbot rebuilds packages also in 19.07, so your package lists get stale.)

closed 2020-06-17 by hnyman

predators46, 2020-06-17:
I use imagebuilder