openwrt/luci #4188: luci-app-wol: feature request: add host bookmarks

lunzet, 2020-06-18:
Hello
could there be a feature like adding a single host as a favourite/bookmark?
scrolling through the list of interfaces + list of hosts in the network is kinda painful when you have a larger set of hosts in the network :-)

Thank you
Thomas

jow-, 2020-06-18:
Uhm... sounds complicated. What about simply rmeembering the last chosen value? Would that work?

lunzet, 2020-06-18:
yeah could work also. just something which gets me quicker to the host i want to wol again. a way to configure which interfaces and which hosts to discplay would also help (so reducing the list items)

AtumVirt, 2021-09-11:
Rather than bookmarks could we simply check/read the /etc/config/etherwake config file?  Targets are defined there as bookmarks already.  Just put the named bookmarks at the top of the list?