openwrt/luci #4195: luci-app-https-dns-proxy: ECS no longer supportedby https-dns-proxy

mikeasolo, 2020-06-21:
Configuring EDNS client subnet" casuses https-dns-proxy to crash with the error message "Unknown state!" and an exit code of 1. 

This is expected behavior per the attached issue after ECS support was removed 

https://github.com/aarond10/https_dns_proxy/issues/76

"This option has been removed from the latest version as it had no effect.

The ECS command-line flag was supported by Google in their JSON API but it was never well supported elsewhere and was necessary there because EDNS could not be expressed via the JSON API. Now that we're using the binary proptocol (RFC8484), you should look to configure that with a recursive DNS resolver.

dnsmasq I believe supports this via the --add-subnet option."
by " aarond10 commented on Feb 22"

feckert, 2020-06-25:
@stangri Can you comment on that?

stangri, 2020-06-25:
> @stangri Can you comment on that?

Confirm broken, I think I should fix it in the init script first and remove the fields from the luci app after.

feckert, 2020-06-30:
@stangri thanks for your work

closed 2020-06-30 by feckert