openwrt/luci #4204: luci-app-openvpn lacks imortant features and has no docs

thebiggestmac, 2020-06-24:
luci-app-openvpn does not have OVPN clients setup (when OpenWRT is a server). You could not add user and export (DL) user profile. PFSense UI could be used as a model, however, it should be simplified (too complicated for OpenWRT).

Also, there is virtually no documentation, especially initial OPENVPN setup.

feckert, 2020-06-25:
Please provide more information. What you want to expect. I am not a openvpn expert!

thebiggestmac, 2020-06-25:
Ok, here are two screenshots from PFSense GUI: adding an user and exporting OVPN config file.
[image]

These two are single page (upper & lower essential halves):
[image]
[image]

Using PFSense GUI you can create user, setting password & additional fields, than export OpenVPN config file. This greatly simplifies OpenVPN deployment when using OpenWRT as an OpenVPN server.

I think we don't need too much fields/forms in OpenWRT, just two (sub)pages are essential: add OpenVPN user with password (or blank) maybe with radio button to push default route, and generate & download user's OVPN profile ( /etc/openvpn/user.ovpn). 

Considering initial setup  I'll write later, since it is not really LUCI's function.
Docs: I've found no docs describing all luci-app-openvpn existing forms/fields, many are not easy to understand. Just a cheatsheet with examples would be great.

TIA